f4exb/sdrangel #390: API example to control SSB modulator/demodulator Δf

drom, 2019-07-26:
Related #382 #388 

Implementing script to lock TX - RX frequency I need:
* to detect the change of Δf of SSB demodulator
* change Δf of SSB modulator

I can't find an example of such API in your scripts.

f4exb, 2019-07-27:
This function gets the `inputFrequencyOffset` key from any channel:
https://github.com/f4exb/sdrangel/blob/master/scriptsapi/freqtracking.py#L151

This function sets the `inputFrequencyOffset` of any channel:
https://github.com/f4exb/sdrangel/blob/master/scriptsapi/freqtracking.py#L96

The complete channel setting is stored in the `TRACKING_DICT` item for simplicity then only the `inputFrequencyOffset` key is changed before sending the PATCH request.

BTW at line 112 it should be an Euclidean division (in Python3) I have to fix this.

drom, 2019-07-27:
Great! Thank you! It works.
Here is exactly what I needed: https://github.com/AI6YP/sdrangel-node/blob/master/bin/transceiver.js

closed 2019-07-27 by drom