f4exb/sdrangel #392: HackRf with Portapack (and Havoc firmware?) incompatibility

Sinager, 2019-07-29:
Hi,
SDRAngel is great and runs well when I flash my HackRf with the standard firmware, however won't be recognized when the optional standalone controller (Portapack) is connected.
While the controller allows to run HackRf as a stock one -and as such is ok with other software- the hackrf_info command gives two slightly different outputs:

stock:
_Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f9b9be1
Board ID Number: 2 (HackRF One)
Firmware Version: git-509c8f4 (API:1.03)
Part ID Number: 0xa000cb3c 0x00484748_

with Portapack
_Found HackRF
Index: 0
Serial number: RunningFromRAM
Board ID Number: 2 (HackRF One)
Firmware Version: git-ef5b907 (API:1.03)
Part ID Number: 0xa000cb3c 0x00484748_

Chances are that SDRAngel doesn't like the absence of a serial number.
Anyone else experienced the same issue?

A mirror issue has been opened on the Havoc firmware GitHub project.

All the best
Stefano IZ0MJE

f4exb, 2019-07-29:
See: https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/hackrfinput/hackrfinputplugin.cpp#L83 for an explanation

f4exb, 2019-07-29:
Issue in Havoc firmware project: https://github.com/furrtek/portapack-havoc/issues/251

Pedro9012, 2019-08-03:
hola , instala microsoft visual c++  todas desde 2012 . a mi me funciona con portapac.
a mi tampoco lo reconocia.
hello, install microsoft visual c ++ all since 2012. It works for me with portapac.

Saludos

Sinager, 2019-08-04:
hello Pedro,
I confirm it runs ok with the standard PortaPack firmware.
It's the "Havoc" version not ok, possibily beacause it returns a string instead of a serial number when listing devices.

ciao

Pedro9012, 2019-08-06:
Tengo firmware havoc  y me va de lujo , pero con firmware 2017 , actualice y no me funcionaba .
no tengo ahora mi hack , pero cuando este en casa en unos dias intentare darte la combinacion de firm que tengo, me acuerdo que tuve que volver a firm mas antiguos para que funcionara ok.

I have havoc firmware and it is fancy, but with firmware 2017, I updated and it didn't work.
I don't have my hack now, but when I'm home in a few days I will try to give you the combination of firm that I have, I remember that I had to re-sign older to work ok.

Saludos.

aneitzel, 2019-11-16:
Hello Pedro, found the solution! 

Flash Portapack HAVOC firmware from https://github.com/furrtek/portapack-havoc/issues/265
It contains a fix for display serial number correctly as SDRangel needs to work with it.

https://www.mediafire.com/file/dekkiftbl8tavch/portapack-h1-havoc-vma.bin/file

For me this PortaPack HAVOC firmware works fine now with SDRangel! :-) You need to test some settings as 
sample rate = 10Mhz
Dezimation = 4
BBF = 20Mhz

those settings works best here
regards André

Pedro9012, 2019-11-24:
Hola Aneitzel, me alegro de que lo hallas conseguido , yo cambie de ordenador y perdi todos los archivos que me ayudaron a ponerlo a punto y ahora mismo no me atreveria a tocar nada de tema firmware , lo hize con una imagen de una version de linux y no me explico como lo consegui.
SALUDOS.
Hi Aneitzel, I'm glad you found it, I changed my computer and lost all the files that helped me get it ready and right now I wouldn't dare to touch any firmware issue, I did it with an image of a Linux version and I don't explain how I got it.
REGARDS.

closed 2019-12-19 by f4exb