f4exb/sdrangel #393: Use WebSocket instead of HTTP/REST + Reverse API

drom, 2019-07-29:
Does it make more sense using single WebSocket over HTTP connection instead of HTTP/REST + Reverse HTTP/REST?
It would be possible then to use Web APP running in the browser to control SDRangel.

f4exb, 2019-07-29:
No sorry this is HTTP REST or nothing. The choice has been made for HTTP REST to control SDRangel and I will not come back on it.

closed 2019-07-29 by f4exb

drom, 2019-07-30:
Does it implement reverse API

drom, 2019-07-30:
I use reverse API only to lock Delta F in SSB Demod to SSB Mod.
https://github.com/AI6YP/sdrangel-node/blob/master/bin/transceiver.js#L73
The reset I can do from the browser.

f4exb, 2019-07-30:
Originally the REST API was developed to control SDRangel from a program outside or from a web application like sdrangelcli.

Then the "reverse" API was implemented in SDRangel for it to control other programs that implement a subset of the same REST API. It uses a subset of the verbs and the same JSON syntax but is limited to device and channel settings and activated individually for each device and channel. This way you can for example have the opportunity (with some custom software in between) to control a filter switchbox depending on the device frequency. 

This does not address web apps in this direction because it simply is not meant to do so. Maybe this is where websockets would fit but I am not going to implement it. This is not a web app and the prime user interface is the Qt GUI.