f4exb/sdrangel #395: Question on device UIs (advanced settings in device UI)

rgetz, 2019-07-29:
Quick question:

I was wondering what you thought of "advanced settings" in device controls. I wanted to add a few more things to the Pluto, but didn't want to make the window longer, and was hoping I could add something like:
[image]

(like in the many of the channel controls). before I went though the effort - thought I would ask if it was a good or bad idea. I think it should just be a matter of setting up an additional *.ui, and defining it as a RollupWidget?

-Robin

f4exb, 2019-07-29:
I think the RollupWidget was made for the purpose of being able to collapse sections so this is probably the way to go.

rgetz, 2019-07-31:
Ok - thanks for the pointer - I will try that, and let you know if I have any questions...

closed 2019-07-31 by rgetz