f4exb/sdrangel #396: Request: documentation for developers

jmi2k, 2019-07-31:
Issue addressing the same problem as #98, but not only for demodulator plugins, but in general for the whole project. Some notes about how to start hacking some plugins, an overview of the structure of SDRangel and maybe one completely documented simple plugin to take as an example would make contributing easier.

They don't have to be very detailed nor very specific, just some pointers to where to start and some advice through the basics of writing plugins, and some notes about the overall architecture of SDRangel.

rgetz, 2019-07-31:
Like any open source project - documentation pull requests are always encouraged. :)

The easiest way is to start looking (or at least how I did it) was to run doxygen on the source to generate all the class listings, and call graphs. Its still missing doc, but the pictures, and understanding what calls what will help alot.

f4exb, 2019-07-31:
I am an advocate of Agile at work and in my personal projects also. One of the core four values of Agile is : "working software over comprehensive documentation". I think is says it all...

rgetz, 2019-07-31:
Having a few autogenerated pictures like:

[image]

or

[image]

and allowing people to browse them (on http://f4exb.github.io/sdrangel/index.html or somewhere else) where they are all clickable - might be worthwhile (or it might generate a bunch of questions you don't want to answer - could go either way) On other projects I'm involved in - we have github pages updated from travis-ci on every release tag. I can do the same here, and send a pull request if you want. Most of the scripts should be exactly the same, you would just need to encrypt a github token that allows a travis-ci build to write into the repo...

Right now, I do run doxygen locally, and it does help from (at least my) newbie perspective. 

I can provide the Doxyfile if you want, so people can generate things themselves in a standard manner. And 5 lines of text in the wiki is also fine - your call. 

-Robin

jmi2k, 2019-07-31:
That's something, and indeed very useful, although I still think some proper documentation somewhere would foster contributions. But I understand that's something you would want to delay until the project is more stable.

@rgetz I'd appreciate anything which could help me get started writing a plugin. I haven't used Doxygen before

f4exb, 2019-08-01:
Sure the Doxyfile could be added in the source tree and a specific build target could generate the files. Also the program itself has a HTTP server mainly used for the API that could serve the files live but I am worried it could make it inflate a lot. Offline doc is always a possibility anyway.

rgetz, 2019-08-01:
OK - feel I will send a pull request with some cmake integration.

rgetz, 2019-08-02:
@jmi2k have a look at : https://rgetz.github.io/sdrangel/ 

Poke through some things - if you are looking new hardware, it would be:
https://rgetz.github.io/sdrangel/dir_06502214778ff2fe8ad4c51ebf4947b1.html

click on the triangle to expand the picture. Click on any block in the picture to explore.
[image]

If you are looking at a modulator, or demod - start looking at those examples.

jmi2k, 2019-08-03:
At least now there is a better way of navigating the code and getting the overall picture of how things are related. Thanks @rgetz

I'm closing this issue for now. The docs generated by the last PR aren't the best solution _possible_, but considering they are generated automatically (so zero cost for maintainers/contributors), more than enough for getting the overall picture. It might be worth it to reopen this issue in the future, but now it's kinda solved pretty nicely so I'll leave that decision to the maintainers.

closed 2019-08-03 by jmi2k

rgetz, 2019-08-03:
@jmi2k if you build the doc yourself - there is a dox.warn left in the build directory, which details all the missing definitions, and document warnings. As you look through hints and get a better understanding of things - Pull requests are always encouraged. ;)

rgetz, 2019-08-05:
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux#developer-docs

f4exb, 2019-08-06:
Looks great thanks! :smile: