f4exb/sdrangel #397: My view on sdrangel

virendera, 2019-08-01:
I playing with sdrangel from past two days to use it with hackrf one after installed it a long time ago (now latest sdrangel) on Windows 10. Here are some following points which I notice and they should be improve to use it usability. Generally I use it for transmit test with hackrf and receive it on RTL-SDR.

My system configuration are as follows
2gb ddr2 ram
2ghz dual core CPU
2tb hard disk

Sdrangel does not have any options to save settings you need to rearrange everything after application startup maybe config file should be used to save it. 

I notice that it's take huge amount of CPU  above 50% and there still no smooth audio transmit. Audio is sometimes choppy or sometime it stuck. Application need to optimize for resource allocation and maximum throughput. Maybe in transmission mode some processing and memory unit should be keep in advance reserve which is minimum requirement for that transmission.

There are lots of modules which are default shows in application and when close them they just hide regarding they should be close. If I just try to use sdrangel for only transmit purpose then there a default rx sink open in application. There is no way to close it.

I think there are lots of features added in the application but optimization work not done and it is necessary now because sdr is extensive CPU consuming application. If I try to open sdr applications one for rx and other for tx then I have to close all the application.

Prior to this I use foobar for tx with hackrf and there is no problem in tx and rx it smooth but as it is just a media player.

At the end I would like to thank to software author for his work to share his work with us.

f4exb, 2019-08-01:
As said earlier if you are not satisfied please submit PRs or even fork it to have it exactly to your own desire. 

It all depends on your particular conditions. I have no such issues on a Linux box and even on a RPi3 (server variant) with armv8 O/S it runs smoothly at ~12% CPU with several receiver streams opened. I find your remarks pretty bold as a first contact and you should have a closer look before emitting such remarks.

BTW 2GB RAM is not enough on Windows where O/S background processes eat up a lot of memory without telling you they do so. 2 GHz dual core CPUs are not enough for a DSP application of any sort where threads need to run effectively concurrently.

Keeping arrangement of windows between sessions have their pros and cons (there was a discussion recently about it). In fact this can be more an annoyance than an advantage so I will leave it like this. This cannot be considered as a big deal.

The decimators and interpolators take the bulk of the processing power and they have been optimized: you probably never had a look at the code. Now if you have ideas for improvement please share your DSP talent with us.

I don't understand what you are trying to say about modules that just hide. When you close a channel or change a device the resources are effectively released. Did you really look at the code?

The way things are organized in the GUI mean you keep a receiving stream a.k.a device set (R0) I do not intend to change this because having an empty shell would look quite weird on a GUI application. If you don't need it just load it with the file input plugin and keep it stopped. The server variant does start with zero device sets but as it is headless this is not an issue. Again you probably did not look hard enough.

If this is all you have to say and do not come with concrete suggestions or PRs to sustain your ideas I will eventually close the issue.

virendera, 2019-08-02:
First of all I would like to say that English is not my native language if you feel "pretentious" then apologise for that and it was not my intention. I am just a normal user of your application not know anything about DSP.

I have full respect for your work but as I stated that there is smooth tx with foobar and rx it with sdr#. Maybe this is due to that sdr# and foober dsp extensions (https://github.com/jocover/foo_hackrf) are written in C# that's why it work without zagginess on my current hardware setup and sdrangel uses Qt for window which is not native application framework for windows application development. I have not tried sdrangel on Linux maybe it is more suitable for Linux environment. 

I don't find it is easier to use waterfall and fft to identify an unknown or weak signal, spectrum monitor is not easy.

Yes, I admit that my current hardware is not sufficient for sdr like stuff (even my phone have 6gb ram with 2.8ghz cpu) but I am not doing too much I am just try to configure my setup to simulate a simple walkie talkies where RTL-sdr use for rx and hackrf for tx.

f4exb, 2019-08-02:
Well if you don't know anything about DSP and you did not read the code how can you say things like "...but optimization work not done"? It is just uninformed, irrespectful for the author (me) and useless if you don't have something better to propose. I think this is not related to language or culture it is just common sense. Moreover there is more than just the source code to optimization; maybe you ignore this as well? The compiler plays a key role and I don't know what MSVC does exactly (who knows?) but I doubt that on the Windows platform you can beat C++ code compiled with MSVC. I have not conducted benchmarks against SDR# and quite frankly I don't care but the few times I had used one and the other for similar tasks I noticed that SDR# would be higher on CPU%. This is not enough to draw conclusions but I am not worried about performance vs SDR#.

I have stated in the first lines of the readme that you need to know the basics of DSP to use this software productively. You need to understand the effects of sampling, decimation, interpolation and if you want to see weak signals on the waterfall the effects of FFT size and averaging. I think you should have all the controls to get the desired effects. Now if you look for a simple SDR with waterfall maybe you should look somewhere else.

You say that foobar does a better job when transmitting with the HackRF. Well I did have a look at the foobar code more precisely at the interpolator. I am not saying it is wrong but at least it does not seem to do things in the canonical way. Normally to perform interpolation you would insert as many zero samples after one input sample to fit the total necessary samples in one output cycle and then you would pass the result through a FIR filter and a proper FIR is costly in terms of CPU cycles. I cannot see anything like that in the foobar code and the filtering effect seems to be done with averaging (not sure it is difficult to read). Well there should be good reasons why the DSP engineers do things the "canonical" way. Sure there are ways to spare CPU cycles but I would be curious to see the result on a lab grade spectrum analyzer. BTW there has been already a round of optimization on the interpolation and now I don't know how to do more. If you say the job is not done then you should at least suggest how it should be done.

Having said that since this ticket does not lead to anything productive I am closing it.

closed 2019-08-02 by f4exb