f4exb/sdrangel #399: min capture rate using Pluto

dj0abr, 2019-08-03:
using Pluto, the minimum sample rate can be set to 2.038.668 S/s. This generates a network load of about 60 Mbit/s, which is too fast for most cases resulting in stuttering.
I have read in the documentation that activating FIR and set it to 2 or 4 would allow to run at slower sample rates. But whatever I set (1,2 or 4) the minimum possible rate is fixed at 2.038.668 S/s, and the network load stays at 60MB/s.
I'm using a Notebook with Intel i7 CPU running under Ubuntu and the latest .deb binary from here.
Pluto's HW sampling rate can be set down to 0,55 MB/s with i.e. SDRconsole (under Windows). What can I do to reduce the HW sampling rate in sdrangel ?
73's Kurt, DJ0ABR

f4exb, 2019-08-04:
Pluto is not supported through network (USB only) so my first question is how do you do achieve this?

f4exb, 2019-08-10:
No answer since 7 days ...

closed 2019-08-10 by f4exb