f4exb/sdrangel #400: DSD Plugin is not loaded

parameter-pollution, 2019-08-04:
I just tried the current (4.11.5) sdrangel for the first time (on Linux Mint 19.2) and I really like it!
But I noticed that when I start it (either via the cinnamon menu or the terminal), the DSD plugin is missing and I found this in the log:
`2019-08-04 12:24:42.970 (W) PluginManager::loadPluginsDir: Cannot load library /usr/lib/sdrangel/plugins/libdemoddsd.so: (libmbe.so.1: cannot open shared object file: No such file or directory)`
When I change my working directory to /usr/lib/sdrangel/ and start it from there then it works.
So I added this line to the .desktop file ( /usr/share/applications/sdrangel.desktop) : 
`Path=/usr/lib/sdrangel/`
and now it works via the cinnamon menu too.
If this is not a general bug and I am the only one affected then feel free to just close it.

f4exb, 2019-08-04:
It is not a bug rather a limitation of the build. See: https://github.com/f4exb/sdrangel#ubuntu
You have to do `export LD_LIBRARY_PATH=/usr/lib/sdrangel` prior to launching sdrangel. Of course if you make `/usr/lib/sdrangel` the current directory it works also but if you write a simple script for startup then the first solution is better.

f4exb, 2019-08-11:
As no action needed I close this issue.

closed 2019-08-11 by f4exb