f4exb/sdrangel #401: Crash w/ maxhold spectrum display and "large" FFT size

eblot, 2019-08-08:
sdrangel crashes with the following setup:

2019-08-08 17:47:20.588 (D) using qt5ct plugin
SDRangel 4.11.5

Linux Mint 19.1 (based on Bionic 18.04 LTS)

- LimeSDR RX0 (Hi antenna)
- LO 2.45GHz
- SR: 2OMS/s
- 4K FFT
- Display Max Hold

The issue can be triggered while maxhold is selected once the SDR is capturing packets, or while it is idle, i.e. start app, selecting LimeSDR and Hi antenna, configure settings, select FFT, select maxhold -> crash, w/o even triggering the "play" button.

The issue seems to be related to the combination 4K FFT/Max hold display.
GDB reports the following backtrace:

    Thread 1 "sdrangel" received signal SIGSEGV, Segmentation fault.
    __memmove_sse2_unaligned_erms ()
        at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:371
    371   ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
    (gdb) bt
    #0  0x00007ffff50a1641 in __memmove_sse2_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:371
    #1  0x00007ffff2d33dc5 in  () at /usr/lib/x86_64-linux-gnu/libGL.so.1
    #2  0x00007ffff2d83815 in glDrawArraysEXT () at /usr/lib/x86_64-linux-gnu/libGL.so.1
    #3  0x00007ffff7aea8e0 in GLShaderSimple::draw(unsigned int, QMatrix4x4 const&, QVector4D const&, float*, int) ()
        at /usr/lib/sdrangel/libsdrgui.so
    #4  0x00007ffff7af2b7b in GLSpectrum::paintGL() () at /usr/lib/sdrangel/libsdrgui.so
    #5  0x00007ffff4dafcc4 in QGLWidget::glDraw() () at /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5
    #6  0x00007ffff4daf66d in QGLWidget::paintEvent(QPaintEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5
    #7  0x00007ffff69bf038 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #8  0x00007ffff698082c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #9  0x00007ffff69880f4 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #10 0x00007ffff5c029a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #11 0x00007ffff69b818a in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #12 0x00007ffff69b874a in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*,     QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #13 0x00007ffff698df4b in QWidgetPrivate::repaint_sys(QRegion const&) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #14 0x00007ffff69a76a2 in QWidgetPrivate::syncBackingStore() () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #15 0x00007ffff69bf1a8 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #16 0x00007ffff698082c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #17 0x00007ffff69880f4 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #18 0x00007ffff5c029a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #19 0x00007ffff5c0511d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
        at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #20 0x00007ffff5c5c2c3 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #21 0x00007ffff16de417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
    #22 0x00007ffff16de650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
    #23 0x00007ffff16de6dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
    #24 0x00007ffff5c5b8ef in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
        at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #25 0x00007ffff5c009ea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #26 0x00007ffff5c09a84 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #27 0x000055555555650c in  ()
    #28 0x0000555555555c16 in main ()

f4exb, 2019-08-12:
I made a quick test but I cannot reach 20 Ms/s without dropping packets. I think I should have the Lime alone on the USB system. Anyway I had a closer look at how `GLSpectrum::paintGL()` works. It starts by applying changes including resizing arrays if the `m_changesPending` boolean is true. Before doing this it makes sure it can acquire the mutex. However in many places the mutex is not taken by the methods that set `m_changesPending` to true which is potentially unsafe.

f4exb, 2019-08-14:
The above mentioned safety measure was implemented in v4.11.6, Do you still encounter the issue?

eblot, 2019-08-19:
Sorry for the delayed response.

I also lose packets.
I just gave a new try w/ 4.11.6, same issue:
2019-08-19 10:30:07.854 (I) SDRangel 4.11.6 Qt 5.9.5 64b x86_64 Linux Mint 19.2 DSP Rx:24b Tx:16b PID 6591

    Thread 1 "sdrangel" received signal SIGSEGV, Segmentation fault.
    __memmove_sse2_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:371
    371   ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
    (gdb) bt
    #0  0x00007ffff5075641 in __memmove_sse2_unaligned_erms () at ../sysdeps/x86_64/multiarch/    memmove-vec-unaligned-erms.S:371
    #1  0x00007ffff2cc1dc5 in  () at /usr/lib/x86_64-linux-gnu/libGL.so.1
    #2  0x00007ffff2d11815 in glDrawArraysEXT () at /usr/lib/x86_64-linux-gnu/libGL.so.1
    #3  0x00007ffff7aeaa90 in GLShaderSimple::draw(unsigned int, QMatrix4x4 const&, QVector4D     const&, float*, int) () at /usr/lib/sdrangel/libsdrgui.so
    #4  0x00007ffff7af30bb in GLSpectrum::paintGL() () at /usr/lib/sdrangel/libsdrgui.so
    #5  0x00007ffff4d83cc4 in QGLWidget::glDraw() () at /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5
    #6  0x00007ffff4d8366d in QGLWidget::paintEvent(QPaintEvent*) () at /usr/lib/x86_64-linux-gnu/    libQt5OpenGL.so.5
    #7  0x00007ffff6993038 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/    libQt5Widgets.so.5
    #8  0x00007ffff695482c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/    x86_64-linux-gnu/libQt5Widgets.so.5
    #9  0x00007ffff695c0f4 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu    /libQt5Widgets.so.5
    #10 0x00007ffff5bd69a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/    x86_64-linux-gnu/libQt5Core.so.5
    #11 0x00007ffff698c18a in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /usr/lib/    x86_64-linux-gnu/libQt5Widgets.so.5
    #12 0x00007ffff698c74a in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint     const&, int, QPainter*, QWidgetBackingStore*) ()
        at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
    #13 0x00007ffff6961f4b in QWidgetPrivate::repaint_sys(QRegion const&) () at /usr/lib/    x86_64-linux-gnu/libQt5Widgets.so.5
    #14 0x00007ffff697b6a2 in QWidgetPrivate::syncBackingStore() () at /usr/lib/x86_64-linux-gnu/    libQt5Widgets.so.5
    #15 0x00007ffff69931a8 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/    libQt5Widgets.so.5
    #16 0x00007ffff695482c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/    x86_64-linux-gnu/libQt5Widgets.so.5
    #17 0x00007ffff695c0f4 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu    /libQt5Widgets.so.5
    #18 0x00007ffff5bd69a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/    x86_64-linux-gnu/libQt5Core.so.5
    #19 0x00007ffff5bd911d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)     () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #20 0x00007ffff5c302c3 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #21 0x00007ffff166c417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/    libglib-2.0.so.0
    ---Type <return> to continue, or q <return> to quit---
    #22 0x00007ffff166c650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
    #23 0x00007ffff166c6dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/    libglib-2.0.so.0
    #24 0x00007ffff5c2f8ef in QEventDispatcherGlib::processEvents(QFlags<    QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    #25 0x00007ffff5bd49ea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/    x86_64-linux-gnu/libQt5Core.so.5
    #26 0x00007ffff5bdda84 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/    libQt5Core.so.5
    #27 0x000055555555650c in  ()
    #28 0x0000555555555c16 in main ()
    (gdb)

eblot, 2019-08-19:
Very same issue with PlutoSDR @ 2.5MS/s

rgetz, 2019-08-21:
I'm not sure I can replicate (tried on today's dev branch, built locally)

My setup is a PlutoSDR running 0.31 firmware, with 2.083336 MSPS:
[image]

and
[image]

runs for a long time - no crashes for me.

cxi-eblot, 2019-08-23:
Maybe it is related to the loss of packets ?
I wish I could rebuild it natively on macOS to give it a try, but it seems the code is hardcoded for Linux, so I have to use a VM.

eblot, 2019-08-23:
Maybe it is related to the loss of packets ?
I wish I could rebuild it natively on macOS to give it a try, but it seems the code is hardcoded for Linux, so I have to use a VM.

f4exb, 2019-08-24:
The trace seems to be related to some kind of race condition as it does not look like it belongs to the main thread. I still don't see why it could happen after the revision of mutex handling. Maybe it could be simplified but I don't think that would change the issue. Also do you still get packet drops with the Pluto @ 2.5 MS/s ?

f4exb, 2020-05-16:
I was never able to reproduce this issue

closed 2020-05-16 by f4exb