f4exb/sdrangel #402: No RF out with HackRF one, spectrum view empty

DK2MZL, 2019-08-08:
I am using rev 4.11.5 on Windows.
There is no spectrum display in the center section, only a white square.
If I resize the windows by sliding the vertical bars the window is not refreshed.
If I click the black square in the Spectrum Display section "clear spectrum histogram" the program crashes.
I cannot have RF power out when transmitting with Hackrf One.
In an older version, the 4.9.1 for example, this was not happening.
I am using Windows 10 64 bit, last update. 
PC: HP Pavilion 15 2014
CPU: AMD A6-5200
Graphics: Radeon HD
RAM: 8GB

f4exb, 2019-08-10:
The "white square" in place of the spectrum display is symptomatic of an issue with OpenGL. The most recent Windows releases are now compiled with MSVC 2017 and the Qt support that goes with it which could in some particular instances have side effects.
  
I have no issues on Windows 10 hence this issue can not be reproduced and is due to your particular setup. So sorry nothing to be done here.

closed 2019-08-10 by f4exb

f4exb, 2019-08-10:
another possibility is that AMD A6-5200 does not support SSE 4.1 instructions for which compatibility is now enforced on Windows builds.