f4exb/sdrangel #403: new install dsdcc fails

imesg, 2019-08-18:
No idea what causes this.
new sdrangel install dsdcc stops:

dt1 /opt/build # cd /opt/build
dt1 /opt/build # git clone https://github.com/f4exb/dsdcc.git
Cloning into 'dsdcc'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 1988 (delta 3), reused 8 (delta 2), pack-reused 1975
Receiving objects: 100% (1988/1988), 24.90 MiB | 1.19 MiB/s, done.
Resolving deltas: 100% (1475/1475), done.
dt1 /opt/build # cd dsdcc
dt1 /opt/build/dsdcc # mkdir build; cd build
dt1 /opt/build/dsdcc/build # cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/dsdcc -DUSE_MBELIB=ON -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so ..
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found LibMbe: /opt/install/mbelib/include, /opt/install/mbelib/lib/libmbe.so
-- Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/build/dsdcc/build
dt1 /opt/build/dsdcc/build # make -j $(nproc) install
Scanning dependencies of target dsdcc
[  6%] Building CXX object CMakeFiles/dsdcc.dir/descramble.cpp.o
[  6%] Building CXX object CMakeFiles/dsdcc.dir/dmr.cpp.o
[  9%] Building CXX object CMakeFiles/dsdcc.dir/dsd_decoder.cpp.o
[ 19%] Building CXX object CMakeFiles/dsdcc.dir/dsd_logger.cpp.o
[ 19%] Building CXX object CMakeFiles/dsdcc.dir/dsd_filters.cpp.o
[ 19%] Building CXX object CMakeFiles/dsdcc.dir/dsd_mbe.cpp.o
[ 22%] Building CXX object CMakeFiles/dsdcc.dir/dsd_opts.cpp.o
[ 25%] Building CXX object CMakeFiles/dsdcc.dir/dsd_state.cpp.o
[ 29%] Building CXX object CMakeFiles/dsdcc.dir/dsd_symbol.cpp.o
[ 32%] Building CXX object CMakeFiles/dsdcc.dir/dstar.cpp.o
[ 35%] Building CXX object CMakeFiles/dsdcc.dir/ysf.cpp.o
[ 38%] Building CXX object CMakeFiles/dsdcc.dir/dpmr.cpp.o
[ 41%] Building CXX object CMakeFiles/dsdcc.dir/nxdn.cpp.o
[ 45%] Building CXX object CMakeFiles/dsdcc.dir/nxdnconvolution.cpp.o
[ 48%] Building CXX object CMakeFiles/dsdcc.dir/nxdncrc.cpp.o
[ 51%] Building CXX object CMakeFiles/dsdcc.dir/nxdnmessage.cpp.o
[ 54%] Building CXX object CMakeFiles/dsdcc.dir/p25p1_heuristics.cpp.o
[ 58%] Building CXX object CMakeFiles/dsdcc.dir/dsd_upsample.cpp.o
[ 61%] Building CXX object CMakeFiles/dsdcc.dir/fec.cpp.o
[ 64%] Building CXX object CMakeFiles/dsdcc.dir/viterbi.cpp.o
[ 67%] Building CXX object CMakeFiles/dsdcc.dir/viterbi3.cpp.o
[ 70%] Building CXX object CMakeFiles/dsdcc.dir/viterbi5.cpp.o
[ 74%] Building CXX object CMakeFiles/dsdcc.dir/crc.cpp.o
[ 77%] Building CXX object CMakeFiles/dsdcc.dir/pn.cpp.o
[ 80%] Building CXX object CMakeFiles/dsdcc.dir/mbefec.cpp.o
[ 83%] Building CXX object CMakeFiles/dsdcc.dir/locator.cpp.o
[ 87%] Building CXX object CMakeFiles/dsdcc.dir/phaselock.cpp.o
make[2]: *** No rule to make target '/opt/install/serialdv/lib/libserialdv.so', needed by 'libdsdcc.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 90%] Building CXX object CMakeFiles/dsdcc.dir/timeutil.cpp.o
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/dsdcc.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

f4exb, 2019-08-18:
This concerns dsdcc really but never mind... My guess is that when it says: `Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so` it may be lying to you. This is because it takes for granted the `-DLIBSERIALDV_INCLUDE_DIR` and `-DLIBSERIALDV_LIBRARY` so please check if the serialdv includes and libs do exist in the folders you specify or if these folders exist at all.

This sort of situation has been fixed in the CMakeLists.txt of sdrangel by forcing it to search for the files with `-DSERIALDV_DIR=/opt/install/serialdv`. Notice that you do not give it the path to includes or the .so library but just the top directory where it should find them. However in dsdcc it is still coded the old way.

imesg, 2019-08-19:
Thanks for the clue. 

dedcc was looking for opt/install/serialdv/lib/ but only lib64 was there. I linked them then dsdcc completes.

closed 2019-08-19 by imesg