f4exb/sdrangel #405: device does FFT?

rgetz, 2019-08-21:
We have some PlutoSDR firmware that does the FFT inside the Pluto's FPGA, -- rather than transferring I/Q samples up it transfers FFT bins.

Is that something that might be useful? (It doubles the effective FFT bandwidth; rather than starting to overflow ~6-7 MSPS; you can now get 12-14 MHz of FFT information without dropping bins. (This assumes that we toss phase information, and only pass back log(magnitude) of the FFT bins.

However - this would completely mess up any other channelizer/decoder, and is really only useful for the main FFT window...

I didn't see the concept of "I/Q samples" or "FFT bins" in the device stream infrastructure - it's really a question of is that something that might ever go into master? or is this too wacky of a use case to support at all?

Thanks

f4exb, 2019-08-22:
As you probably guessed SDRangel takes the strong assumption that the input and also output are I/Q samples. It is not totally impossible to imagine something else but it would have to go in a new different flow (not sample source nor sample sink). 

I have started the design of a mixed flow (called MIMO) to handle multiple Rx and Tx streams at the same time (still I/Q) which makes the core of the "v.5 project". This is to support the coherent processing of multiple input/outputs for interferometry, beam-forming, radar and the like so in most places I already turned the Rx/Tx boolean to a more generic enum and generalized things a bit more to allow more different schemes of data flow. This is why I say it is not totally impossible but requires a bit of thinking...

donzelot, 2020-02-24:
> We have some PlutoSDR firmware that does the FFT inside the Pluto's FPGA, -- rather than transferring I/Q samples up it transfers FFT bins.
>
> Is that something that might be useful? (It doubles the [...]

Dear All,
Is that firmware with FPGA processed FFT available and where ?
I was not able to find it anywhere.
Same question for RFNOC like accelerated FFT for gr-fosphor.

Thanks
Chris

f4exb, 2020-05-27:
Eventually this is not going to happen because there is nothing else useful to do other than display with just the FFT bins as stated in this phrase:

> However - this would completely mess up any other channelizer/decoder, and is really only useful for the main FFT window...

Moreover I suppose this is the PSD samples not the full complex FFT samples so there is no way to recover the complex signal.

closed 2020-05-27 by f4exb