f4exb/sdrangel #406: Crash when toggling between 2.4k and 4.8k DSD

jigpu, 2019-08-22:
Switching back and forth between 2.4k and 4.8k will sometimes cause the program to crash. The cause appears to be an uncaught floating-point exception within the DSD library:

~~~
Thread 26 "QThread" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fff990bd700 (LWP 6826)]
0x00007fffc5ad19ba in DSDcc::DSDSymbol::pushSample (this=this@entry=0x555557f564d8, sample=sample@entry=0) at /usr/src/debug/dsdcc/dsd_symbol.cpp:265
265	/usr/src/debug/dsdcc/dsd_symbol.cpp: No such file or directory.
(gdb) bt full
#0  0x00007fffc5ad19ba in DSDcc::DSDSymbol::pushSample(short) (this=this@entry=0x555557f564d8, sample=sample@entry=0) at /usr/src/debug/dsdcc/dsd_symbol.cpp:265
#1  0x00007fffc5ace430 in DSDcc::DSDDecoder::run(short) (this=0x555557f47770, sample=<optimized out>) at /usr/src/debug/dsdcc/dsd_decoder.cpp:378
#2  0x00007fffc5b6443b in DSDDecoder::pushSample(short) (sample=<optimized out>, this=0x555557f47770) at /home/jgerecke/.cache/yay/sdrangel-git/src/sdrangel/plugins/channelrx/demoddsd/dsddecoder.h:32
        sample = 0
        sampleDSD = <optimized out>
        im = <optimized out>
        magsq = <optimized out>
        demod = <optimized out>
        delayedSample = <optimized out>
        re = <optimized out>
        it = {m_real = 0, m_imag = 0}
        ci = {_M_value = -304.602631 + -148.721497 * I}
        samplesPerSymbol = 10
#3  0x00007fffc5b6443b in DSDDemod::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x555557f47490, begin=..., end={m_real = 620, m_imag = -409}, firstOfBurst=<optimized out>)
    at /home/jgerecke/.cache/yay/sdrangel-git/src/sdrangel/plugins/channelrx/demoddsd/dsddemod.cpp:218
        sample = 0
        sampleDSD = <optimized out>
        im = <optimized out>
        magsq = <optimized out>
        demod = <optimized out>
        delayedSample = <optimized out>
        re = <optimized out>
        it = {m_real = 0, m_imag = 0}
        ci = {_M_value = -304.602631 + -148.721497 * I}
        samplesPerSymbol = 10
#4  0x00007ffff78d8934 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x555558a71750, begin={m_real = 1251, m_imag = 2430}, end={m_real = 2885, m_imag = -3236}, positiveOnly=<optimized out>)
    at /usr/include/c++/9.1.0/bits/stl_iterator.h:871
#5  0x00007ffff79232a4 in ThreadedBasebandSampleSinkFifo::handleFifoData() (this=0x555558656fc0) at /usr/include/c++/9.1.0/bits/stl_iterator.h:871
        part2begin = {m_real = 186, m_imag = -2950}
        part2end = {m_real = 186, m_imag = -2950}
        count = <optimized out>
        part1begin = {m_real = 1251, m_imag = 2430}
        part1end = {m_real = 2885, m_imag = -3236}
#6  0x00007ffff692744a in QObject::event(QEvent*) () at /usr/lib/libQt5Core.so.5
#7  0x00007ffff72b04d5 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#8  0x00007ffff72b9a21 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#9  0x00007ffff68fa9c2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#10 0x00007ffff68fd739 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/libQt5Core.so.5
#11 0x00007ffff69533a4 in  () at /usr/lib/libQt5Core.so.5
#12 0x00007ffff4ae2cf4 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#13 0x00007ffff4ae4b11 in  () at /usr/lib/libglib-2.0.so.0
#14 0x00007ffff4ae4b51 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#15 0x00007ffff69529a3 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#16 0x00007ffff68f95ec in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#17 0x00007ffff672c2f5 in QThread::exec() () at /usr/lib/libQt5Core.so.5
#18 0x00007ffff672d520 in  () at /usr/lib/libQt5Core.so.5
#19 0x00007ffff563b57f in start_thread () at /usr/lib/libpthread.so.0
#20 0x00007ffff63a00e3 in clone () at /usr/lib/libc.so.6
~~~

Looking more closely, it seems that neither the `m_count` nor `m_sum` variables were ever incremented, resulting in the DSD library trying to calculate 0/0. I don't quite follow the code, so I'm not sure under what condition this might be triggered.

~~~
(gdb) p m_count
$1 = 0
(gdb) p m_sum
$2 = 0
(gdb) p m_sampleIndex
$3 = 9
(gdb) p m_samplesPerSymbol
$4 = 10
~~~

**OS:** Arch Linux
**SDRangel Commit:** ec22056cb
**DSDcc Commit:** a0f4694

f4exb, 2019-08-22:
I am thinking of a race condition between the thread that does the baud rate switch and the one running the decoder.

f4exb, 2019-08-31:
Implemented in v4.11.7

closed 2019-08-31 by f4exb