f4exb/sdrangel #408: Nixos no sound

Alkeryn, 2019-08-22:
So i finally succesfuly built it and have it work with my limesdr, it dosen't see my audio output devices tho so i don't have any sound, any idea why that is ?
maybe some compilation flags or libs ?

f4exb, 2019-08-22:
That -1 Hz sample rate is strange. It should come from the config file normally and that may have been accidentally corrupted. You can try to erase `~/.config/f4exb/SDRangel.conf` so it will rebuild it from scratch.

Alkeryn, 2019-08-22:
I tried removing it, changed nothing ^
Also i rebuild the whole thing with libav because it was missing, didn't fixed it ^
it just looks like that : 
[image]

f4exb, 2019-08-22:
What does the command `pacmd list-sinks` return?

Alkeryn, 2019-08-22:
```
❯ pacmd list-sinks
1 sink(s) available.
  * index: 0
	name: <alsa_output.pci-0000_00_1f.3.analog-stereo>
	driver: <module-alsa-card.c>
	flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
	state: RUNNING
	suspend cause: (none)
	priority: 9039
	volume: front-left: 18866 /  29% / -32,45 dB,   front-right: 18866 /  29% / -32,45 dB
	        balance 0,00
	base volume: 65536 / 100% / 0,00 dB
	volume steps: 65537
	muted: no
	current latency: 178,66 ms
	max request: 36 KiB
	max rewind: 36 KiB
	monitor source: 2
	sample spec: s16le 2ch 44100Hz
	channel map: front-left,front-right
	             Stéréo
	used by: 1
	linked by: 1
	configured latency: 210,00 ms; range is 0,50 .. 371,52 ms
	card: 3 <alsa_card.pci-0000_00_1f.3>
	module: 9
	properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "ALC1220 Analog"
		alsa.id = "ALC1220 Analog"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0xde140000 irq 178"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1f.3"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "a2f0"
		device.product.name = "200 Series PCH HD Audio"
		device.form_factor = "internal"
		device.string = "front:0"
		device.buffering.buffer_size = "65536"
		device.buffering.fragment_size = "32768"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-stereo"
		device.profile.description = "Stéréo analogique"
		device.description = "Audio interne Stéréo analogique"
		alsa.mixer_name = "Realtek ALC1220"
		alsa.components = "HDA:10ec1220,18491221,00100003 HDA:8086280b,80860101,00100000"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	ports:
		analog-output-lineout: Sortie ligne (priority 9900, latency offset 0 usec, available: yes)
			properties:
				
		analog-output-headphones: Casque audio (priority 9000, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "audio-headphones"
	active port: <analog-output-lineout>
```

f4exb, 2019-08-22:
So it appears that you do have pulseaudio and it should work but for some reason Qt does not recognize it. In the log of SDRangel near the start you should see a line like (you may need to turn on Debug level for the log):
```
...
2019-08-22 16:30:17.207 (D) AudioDeviceManager::AudioDeviceManager: input device #18: alsa_output.pci-0000_00_1f.3.analog-stereo
```
The number here is arbitrary the point is to see that pulseaudio device. I guess this is not the case since it should appear in the list in the GUI as well and as the system device.

Alkeryn, 2019-08-22:
the only line there is is this one ^ : 
```
2019-08-22 15:40:32.105 (I) SDRangel 4.11.6 Qt 5.12.0 64b x86_64 NixOS 19.09pre186857.239fffc90d7 (Loris) DSP Rx:24b Tx:16b PID 24748
```

f4exb, 2019-08-22:
Turn Debug on!

Alkeryn, 2019-08-22:
I did (if you mean logging debug in the ui) or is there a command line flag for debug ?
or do you want me to compile with debug flag ?

Alkeryn, 2019-08-22:
Oh that one is new : 
```
2019-08-22 17:15:13.827 (W) AudioDeviceManager::setOutputDeviceInfo: index: -1 device: System default device no FIFO registered yet hence no audio output has been allocated yet
```

f4exb, 2019-08-22:
The issue with debug is that Qt turns it off if you don't specify -DDEBUG_OUTPUT=ON

In the main CMakeLists.txt default is OFF:
```
option(DEBUG_OUTPUT "Print debug messages" OFF) 
...
if(NOT DEBUG_OUTPUT)
  add_definitions("-DQT_NO_DEBUG_OUTPUT")
endif()
```
So my guess is that you need that option on your cmake command line.

Alkeryn, 2019-08-22:
Oh ok, recompilling it ^

Alkeryn, 2019-08-22:
i don't think there is anything useful but here /shrug
[image]

f4exb, 2019-08-22:
Strangely enough when debug is working you should see some messages between the program start at ~17:40 and the main window close event at ~17:41. It seems the debug flag was not taken into account before you closed the main window.

Alkeryn, 2019-08-22:
yeah, idk what i should do ^
it is weird, the only software that does that, i mean audacity or all other softwares have no such issue
so maybe i'm forgetting to had a lib when building idk ^

f4exb, 2019-08-22:
In my opinion if Qt has trouble finding the audio device then it has something to do with some Qt dependencies.

Alkeryn, 2019-08-22:
[image]
Hey i just fixed it !
it was indeed a qt issue and me being new to the nix programming language ^

Alkeryn, 2019-08-22:
Well i do still have some weird audio stutering and can't hear much but at least it detects my devices now ^

Alkeryn, 2019-08-22:
Oh it looks like it is just overloading
like if i reduce S/S to 5M it works
tho on gqrx i can go up to 25M/s and it don't stutter
any recomendation for that ?

f4exb, 2019-08-22:
Is this LimeSDR? I can go up to 10 MS/s without packet drops. It still goes beyond that point without guarantee of signal integrity.

Alkeryn, 2019-08-22:
Huh that's weird on gqrx i can go up to 30M/s
Also weirdly enough disabling the top graph allow me to go up to 20M instead of 8
so it looks like it is cpu bound somehow :thinking:

closed 2019-08-23 by Alkeryn