f4exb/sdrangel #409: issue starting sdrangel 4.11.6-win64.exe on windows 7

jcfpi, 2019-08-22:
App crash, see the attached files. Fresh Installation yersteday with sdrangel-4.11.6-win64.exe. No issue with 4.3.0  build qt 5.11.1 32 bits on windows 7
 Thank you
[image]
[image]

f4exb, 2019-08-22:
What is that you don't understand in sdrangel-4.11.6-**win64**.exe?

f4exb, 2019-08-23:
Windows 32 bits unsupported

closed 2019-08-23 by f4exb

Vort, 2019-08-23:
> What is that you don't understand in sdrangel-4.11.6-**win64**.exe?

It's you, who don't understand that 32-bit system can't have Exception Offset of 0000000000009ff4.

f4exb, 2019-08-23:
True but it does not really matter as I am planning no action on this.