f4exb/sdrangel #415: ERROR / This application failed to start because no Qt platform...

F4GXS, 2019-09-15:
Bonjour Edouard, je viens tous juste d'installé votre logiciel SDRAngel sur mon PC pour pouvoir l'utilisé avec mon HackRF One, j'ai essayé de le démarré et j'ai eu un message d'erreur, le voici

_"sdrangel : This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Available platform plugin are: direct2d, windows."_

J'ai donc réinstallé le logiciel comme indiqué et j'ai toujours la même erreur, pas moyens de faire démarré le logiciel.

Que ce passe t'il ? avez vous une idée ?

73 de F4GXS

f4exb, 2019-09-16:
Firstly please express yourself in English as this is an international community and English is likely to be understood by most of us.

Secondly "This application failed to start because no Qt platform plugin could be initialized" should be a clear enough message so you need to sort out issues with Qt support on your machine first. "Available platform plugin are: direct2d, windows." may suggest this is an issue with graphics. On Windows Qt uses the so called "Angle" for OpenGL support that uses DirectX in the background.

F4GXS, 2019-09-16:
OK

Firstly, I still do not understand where the problem comes from, you tell me that I have to sort out issues with "Qt" support, that's why I ask you how to help me to do that, me and the "Qt" it is not my cup of tea.

Secondly, I think my laptop is from a different age, it still runs under Vista.

f4exb, 2019-09-16:
Qt is none of my business. I am also a Qt user.

F4GXS, 2019-09-16:
business is business.
alright, leave it, i'm go to update my Vista laptop to W10.

Thanks

closed 2019-09-16 by F4GXS