f4exb/sdrangel #417: Device sets other than R0 disappear when program is shut down

goscickiw, 2019-09-21:
When I add another device set, for example T1, and shut down the program, the next time I launch it T1 is gone and I have to set it up again. Also R0 loses all settings except the device selected under Sampling devices control.
I tried using presets, but for sets other than R0 I still have to manually add the set and select the device.
I don't know if it's an issue with the program, a missing feature or maybe I'm doing something wrong?

satfan52, 2019-10-08:
I guess the addition of the specific devices-sets and specific channels could be defined in a custom python script (use case) using the REST API  (the script would have to be launched manually ? automatically ?) after SdrAngel starts. However it would indeed be much more convenient for users which not deep into python script programming if SdrAngel could just imply remember all the device-sets and channels so they do not have to be entered at after every start-up.

f4exb, 2019-10-13:
This is working as designed and this design is not going to change. Making the complete current configuration persistent is in fact an annoyance because you do not necessarily want to reproduce always the same setup. So it can be nice for you but not more generally suitable. The GUI version starts with a Rx on tab 0 this is there only so that it does not look weird on the screen but the server version starts without any device set up so even this is debatable.

The presets store data for more than one device so you have to use them in this sequence:
  1. Load the device into the device set
  2. Load the preset  

That's right the REST API is there to cover any other case and gives you maximum flexibility. In the last version (v4.11.10) the `config.py` script that was in the API examples section has been elevated in the more "official" `scriptsapi` folder and the format of the JSON file has been modified to be more informative (key/value pairs instead of positional data).

This script can be used in a "command" and you can just play this command from the GUI. So I think everything is in place to speed up initial configuration.

closed 2019-10-13 by f4exb