f4exb/sdrangel #418: Publish on Flathub

AsciiWolf, 2019-09-24:
It would be great if the SDRangel [Flatpak](https://github.com/f4exb/sdrangel/tree/master/flatpak) could be finished and published on [Flathub](https://flathub.org) if possible. There is currently no SDR software available on Flathub so having one there would be great.

AsciiWolf, 2019-10-22:
The [SDRangel Flatpak](https://github.com/f4exb/sdrangel/tree/master/flatpak) was finished and should now be working fine (except the [broken waterfall](https://github.com/f4exb/sdrangel/issues/423) on Wayland). It would be great if someone could send a PR to the Flathub Git repo. I unfortunately maintain too many other Flatpak packages so I will probably leave this to someone else.

f4exb, 2020-03-01:
This "soemone else" cannot be me (the Maintainer) thus as mentioned in the Wiki Flatpak is not supported: https://github.com/f4exb/sdrangel/wiki/Quick-start#snap-flatpack-and-the-like

closed 2020-03-01 by f4exb