f4exb/sdrangel #423: Waterfall is missing when running on Wayland

AsciiWolf, 2019-10-07:
I have originally reported it in my Flatpak PR, but I have found that this issue is actually caused by Wayland and happens on a non-Flatpak build as well.

The waterfall part of the SDRangel window is missing/transparent when running on GNOME Shell with Wayland:
[image]

It works fine after switching to X.Org session.

I am using Fedora 30 with GNOME 3.32 and Intel GPU.

f4exb, 2019-10-08:
I have no idea about what Wayland is but it is not handling OpenGL properly or in an expected way. The absense of spectrum is symptomatic of a failing OpenGL initialization. The same thing should occur with all the OpenGL widgets: TV screen, DSD symbol scope, ... Most probably Wayland and Qt OpenGL are not compatible.

AsciiWolf, 2019-10-11:
Hmm, weird. I have no issues with other Qt and OpenGL apps or games.

AsciiWolf, 2019-10-21:
I tried investigating the issue, but no luck. I am not an expert on Qt, but from what I was able to find, OpenGL Qt widgets should work fine on Wayland.

f4exb, 2019-10-22:
We have to conclude Wayland is not compatible with SDRangel and so are other OpenGL applications like some games. Fedora has adopted Wayland as its default which may explain the issue when running on Fedora. We'll see if Wayland gets more adoption and if we'll face issues with the coming Ubuntu 20.04.

See: https://mybroadband.co.za/news/software/294624-after-more-than-10-years-in-development-wayland-still-hasnt-replaced-x-org.html

In the meantime this issue can be closed.

AsciiWolf, 2019-10-22:
The waterfall not working on Wayland seems to be a bug in SDRangel code that happens when running on Wayland, not a Wayland issue. Wayland nowadays is perfectly compatible with 99% of apps and games, almost all modern GTK/Qt apps and many games can run natively on Wayland, the rest is automatically run using [XWayland](https://wayland.freedesktop.org/xserver.html#heading_toc_j_0) compat layer and works fine with no visible performance degradation. I have GNOME with Wayland on all my computers and there are absolutely no issues. Even some legacy software like UT2004 or Baudline runs fine. Wayland is being adopted by more and more distributions and desktop environments. For example, RHEL has Wayland used by-default since version 8.0 and Debian has it as default since version 10. GNOME already deprecated X.Org and develops all the new features primarily for Wayland. KDE is also shifting its development focus to Wayland and deprecation of X.Org. MATE and Cinnamon plan to do the same thing. There are also some projects like [Sway](https://swaywm.org) and other smaller DEs and WMs. Not supporting Wayland nowadays is similar to, for example, not supporting PulseAudio. Wayland is becoming the future of Linux desktop, on the other hand, X.Org is dying and didn't see much development in the last years. Please, consider reopening this issue until the bug in SDRangel is fixed. Thanks!

f4exb, 2019-10-23:
This seems to be a pretty good and balanced summary of Wayland vs X.Org: https://www.reddit.com/r/linux/comments/7z9hoa/on_the_state_of_wayland/

Please note: "Migrating to Wayland requires all applications to be rewritten?
No, it requires everything that does X11 calls to be changed to use Wayland instead. In these times that means mostly that only the toolkits (GTK+, Qt, etc.) must be changed to be compatible with Wayland. The applications then do not need to be touched at all."

This is exactly what I meant. Application programmers should not be concerned at all.

AsciiWolf, 2019-10-23:
Hm, yeah, this may be a Qt toolkit regression after all. Although, it is weird that I use many Qt apps including ones with OpenGL stuff and they all work fine. I will try to look into this when I have more time. However, it would be nice to keep this ticked opened to track the issue, even if it is not an issue directly in the SDRangel code.

f4exb, 2019-10-23:
So be it

chron-isch, 2019-12-03:
@AsciiWolf I'm having the same problem with ArchLinux/sway.
As a workaround you can start with `QT_QPA_PLATFORM=xcb sdrangel` and it forces qt to use the xwayland backend.
[image]

chron-isch, 2020-05-26:
@AsciiWolf  or if you don't want to use xwayland you can also start with `QT_WAYLAND_DISABLE_WINDOWDECORATION=1 sdrangel`

closed 2020-11-05 by f4exb

r3k2, 2023-12-13:
I am having this same issue, considering all Linux distros are officially moving to Wayland, forcing an app to run on non-Wayland is *not* a solution in almost 2024. ( I am using the arch linux aur sdrangel-git

PR #685 (2020-11-05): Workaround Flatpak issues on Wayland
Untested, but should resolve #423.