f4exb/sdrangel #429: Tx Audio input issues

goscickiw, 2019-10-13:
I have encountered some issues with transmitting using Audio input, these issues seem to appear only when Audio input is enabled.

(Audio is recorded with RTL-SDR connected to HackRF output, provided in zip files because github doesn't support audio attachments)

For comparison, this is a 200hz tone from SDRangel's own generator - here everything is good: [200hz_sdrangel.zip](https://github.com/f4exb/sdrangel/files/3721613/200hz_sdrangel.zip)
[image]

When I enable Audio input, no matter what the input device is (here using VB-Cable), a noise of unknown origin can appear. This noise is visible on SDRangel waterfall so I guess it is created before the data is sent to the HackRF. This is how it sounds and looks with no sound provided to the input: [audioinput_noise.zip](https://github.com/f4exb/sdrangel/files/3721619/audioinput_noise.zip)
[image]

Here is a 200Hz tone provided via Audio input, with the noise: [200hz_audioinput_withnoise.zip](https://github.com/f4exb/sdrangel/files/3721624/200hz_audioinput_withnoise.zip)
[image]

I can get rid of the noise if I change some modulator settings to different values, and then back to the values that I want to use. Unfortunately, after this there are still some crackling and popping sounds: [200hz_audioinput.zip](https://github.com/f4exb/sdrangel/files/3721626/200hz_audioinput.zip)
[image]

Sometimes, the noise appears differently, with a crackling sound: [audioinput_differentnoise.zip](https://github.com/f4exb/sdrangel/files/3721652/audioinput_differentnoise.zip)
[image]

I can get rid of this noise as well, but after the procedure, a popping sound is present even when no audio is provided on the input: [audioinput_afterdifferentnoise.zip](https://github.com/f4exb/sdrangel/files/3721656/audioinput_afterdifferentnoise.zip)
[image]

I have tested this on two different computers, my desktop PC with an Intel Core i7-4770 on Windows 10 and my laptop with an Intel Core i5-8250U on Windows 10 and Ubuntu 18.04 (dual boot, not using VM) with the same results.

f4exb, 2019-10-22:
The second test shows noise picked up from the microphone or line. It is more than 40dB down from the carrier so I would not consider this.

The most interesting phenomenon is the "popping" or "cracking" noise (very prominent in the last case) because it can be symptomatic of a FIFO issue. I could not reproduce the case even at 20 MS/s on the HackRF (on Ubuntu 18.04) and any combination of interpolation however the way the FIFOs are handled on the Tx side deserves some careful inspection. I am not sure this can fix this particular problem but having a second look years after it was initially designed makes me perplexed. In particular the SampleSourceFifo and the way it is handled in DSPDeviceSinkEngine looks weird to me now. It does not seem to handle the fact that Qt events may not follow the pace of the hardware as the sample rate increases. The pace at which HackRF wants to pull samples may be irregular in bursts and may depend on particular conditions (what happens on the USB bus) and could be different in different environments which may explain that I could not reproduce the issue. 

This is properly handled on the Rx side but does not seem to be the case for Tx. Moreover the design seems to be overly complicated possibly to circumvent some experimental issues due to the improper handling of the FIFO.

However this is a big architectural change and deserves attention. I have designed a different kind of  FIFO for the need of MIMO (MO) work that I may want to try first before retrofitting in the standard Tx process.

goscickiw, 2019-10-22:
In these tests, I wasn't using a microphone or any other hardware input. I was using VB-Cable which is a virtual device and it shouldn't be possible for it to generate noise on it's own. When I record audio from VB-Cable with any other software (for example Audacity) there is no noise present.

Also if the noise really is originating in the input device, then I shouldn't be able to get rid of it just by temporarily changing SDRangel's modulator settings to a different value... Also when I'm providing an irregular audio signal like speech or music, sometimes the noise can randomly disappear and reappear. I will provide a sample of this when I'm be able to.

f4exb, 2019-10-22:
Fair enough but I do not use a VB-cable on Ubuntu and I am unable to materialize the error as you describe it.

f4exb, 2019-10-23:
Also can you please stick to issues happening in your Ubuntu 18.04 system as I am not able to give much support for Windows.

goscickiw, 2019-10-23:
On Ubuntu I have tested it with a microphone and it behaves in the exact same way - I'm able to remove the noise by changing the WFM modulator RFBW to 80kHz, then to 100, and then back to 125 (If I change from 80 right back to 125 the noise re-appears).
The level of actual noise coming from the microphone is much lower.

closed 2019-12-19 by f4exb