f4exb/sdrangel #430: MacOS Build Errors

ktoksoz, 2019-10-13:
When building in macOS Catalina (10.15) I encountered with following errors:

```
/Users/koray/Projects/HAM/sdr/sdrangel/sdrangel/sdrbase/ambe/ambeengine.cpp:141:26: error: variable has incomplete type 'struct serial_struct'
    struct serial_struct serinfo;
                         ^
/Users/koray/Projects/HAM/sdr/sdrangel/sdrangel/sdrbase/ambe/ambeengine.cpp:141:12: note: forward declaration of 'serial_struct'
    struct serial_struct serinfo;
           ^
/Users/koray/Projects/HAM/sdr/sdrangel/sdrangel/sdrbase/ambe/ambeengine.cpp:154:27: error: use of undeclared identifier 'TIOCGSERIAL'
            if (ioctl(fd, TIOCGSERIAL, &serinfo) == 0)
                          ^
/Users/koray/Projects/HAM/sdr/sdrangel/sdrangel/sdrbase/ambe/ambeengine.cpp:157:37: error: use of undeclared identifier 'PORT_UNKNOWN'
                if (serinfo.type != PORT_UNKNOWN) {
                                    ^

```
[image]

rgetz, 2019-10-13:
The readme states:

The platform of choice to run SDRangel is definitely Linux

Which means - most (all) the developers use Linux and don’t have access to MAC. Well formatted patches are always reviewed if you have the time to look into things in more detail.

f4exb, 2019-10-15:
That's perfectly right however at some point before the redesign of the AMBE engine it used to compile for Mac so I will try to restore the situation by skipping the "linuxities" introduced if compiled with `__APPLE__`. 

Now on dev branch there should be a version that compiles with Mac.

f4exb, 2019-10-23:
Should be fixed in v4.11.11

closed 2019-10-23 by f4exb