f4exb/sdrangel #435: KiwiSDR plugin doesn't work

jmceara, 2019-10-30:
I'm new to SDRAngel, but I think KiwiSDR is not working. I'm using latest version of KiwiSDR (1.334) and SDRAngel (4.11.11) and when I try to connect, I just see "Connecting..." and after a while, "Disconnected" in pinky, nothing more.
I'm using my local KiwiSDR and there's no other user connected/using my kiwi. I've tested in brownser, using same IP address, and everything is working fine in brownser.
Is there any way to enable debug or anything else?
I'm using this url (with my local IP): 192.168.22.168:8073

For instance, my public kiwi address is skynet.jonis.com.br:8073

Tks

jmceara, 2019-10-30:
Not sure if it helps, but I found some logs in KiwiSDR about SDRAngel connection:
`Wed Oct 30 00:32:25 00:01:02.142 0.2..... 0        TLIMIT exempt local connection from 192.168.22.137
Wed Oct 30 00:32:25 00:01:02.142 0.2..... 0        PWD kiwi SND ALLOWED: config pwd set, but is_local and auto-login set
Wed Oct 30 00:33:34 00:02:11.795 0.2..... 0        TLIMIT exempt local connection from 192.168.22.137
Wed Oct 30 00:33:34 00:02:11.795 0.2..... 0        PWD kiwi SND ALLOWED: config pwd set, but is_local and auto-login set
Wed Oct 30 00:33:59 00:02:35.959 0.2..... 0        "(no identity)" 192.168.22.137 incomplete connection kicked`

Vort, 2019-12-28:
Can't connect to your server:
[image]

f4exb, 2020-02-29:
The above shows it was a Kiwi SDR setup issue.

azwirko, 2020-10-15:
Can someone comment on what the KiwiSDR setup issue was that resolved this issue? I am having the same problem using KiwiSDR v1.414 - Essentially the KiwiSDR is dropping the connections because of "no identity" yet I find no settings in KiwiSDR related to ignoring identity.  

`
Thu Oct 15 20:30:11 3d:13:37:22.317 01234...    3     TLIMIT exempt local connection from 192.168.2.204
Thu Oct 15 20:30:11 3d:13:37:22.317 01234...    3     PWD kiwi SND ALLOWED: config pwd set, but is_local and auto-login set
Thu Oct 15 20:30:34 3d:13:37:44.631 01234...    3     "(no identity)" 192.168.2.204 incomplete connection kicked
`

73

andyz - K1RA
https://k1ra.us/

Vort, 2020-10-16:
There was no server for me to make tests.
Now I will look at http://kiwisdr.k1ra.us:8073 and try to find where is the problem.

Vort, 2020-10-16:
@azwirko I don't see any problems with `sdrangel-4.20.0-win64.exe`:
[image]

azwirko, 2020-10-16:
I too am running 4.20.0 64-bit on Windows. Very interesting - looking at the log I can see you connected fine, but when I try to connect locally via the private IP address OR the kiwisdr.k1ra.us name, my connection is incomplete and I am kicked out

`Fri Oct 16 04:48:18 3d:21:55:29.094 01...... 0        TLIMIT-IP connecting LIMIT OKAY cur:1 < lim:360 for 109.87.25.148
Fri Oct 16 04:48:18 3d:21:55:29.095 01...... 0        PWD kiwi SND ALLOWED: no config pwd set, allow any (109.87.25.148)
Fri Oct 16 04:48:34 3d:21:55:45.002 01...... 0          600.00 kHz  iq z0  "(no identity)" 109.87.25.148 (ARRIVED)
Fri Oct 16 04:48:49 3d:21:55:59.643 012.....   2      TLIMIT-IP connecting LIMIT OKAY cur:2 < lim:360 for 109.87.25.148
Fri Oct 16 04:48:49 3d:21:55:59.644 012.....   2      PWD kiwi SND ALLOWED: no config pwd set, allow any (109.87.25.148)
Fri Oct 16 04:48:54 3d:21:56:05.357 .1...... 0          560.00 kHz  iq z0  "(no identity)" 109.87.25.148 (LEAVING after 0:00:36)
Fri Oct 16 04:48:56 3d:21:56:07.307 01...... 0        TLIMIT-IP connecting LIMIT OKAY cur:2 < lim:360 for 109.87.25.148
Fri Oct 16 04:48:56 3d:21:56:07.307 01...... 0        PWD kiwi SND ALLOWED: no config pwd set, allow any (109.87.25.148)
Fri Oct 16 04:48:57 3d:21:56:07.826 01...... 0        PWD kiwi W/F ALLOWED: no config pwd set, allow any (109.87.25.148)
Fri Oct 16 04:49:02 3d:21:56:13.080 01...... 0         7312.82 kHz amn z1  "109.87.25.148" (ARRIVED)
Fri Oct 16 04:49:12 3d:21:56:22.634 .1...... 0         7315.00 kHz amn z7  "109.87.25.148" (LEAVING after 0:00:15)
Fri Oct 16 04:49:15 3d:21:56:25.758 01...... 0        TLIMIT-IP connecting LIMIT OKAY cur:2 < lim:360 for 109.87.25.148
Fri Oct 16 04:49:15 3d:21:56:25.758 01...... 0        PWD kiwi SND ALLOWED: no config pwd set, allow any (109.87.25.148)
Fri Oct 16 04:49:30 3d:21:56:41.041 01...... 0         7315.00 kHz  iq z0  "(no identity)" 109.87.25.148 (ARRIVED)
Fri Oct 16 04:49:48 3d:21:56:59.299 .1...... 0         7315.00 kHz  iq z0  "(no identity)" 109.87.25.148 (LEAVING after 0:00:34)
Fri Oct 16 04:50:03 3d:21:57:13.804 01...... 0        TLIMIT-IP connecting LIMIT OKAY cur:3 < lim:360 for 109.87.25.148
Fri Oct 16 04:50:03 3d:21:57:13.804 01...... 0        PWD kiwi SND ALLOWED: no config pwd set, allow any (109.87.25.148)
Fri Oct 16 04:50:18 3d:21:57:29.022 01...... 0         7315.00 kHz  iq z0  "(no identity)" 109.87.25.148 (ARRIVED)
Fri Oct 16 04:50:53 3d:21:58:04.390 .1...... 0         7315.00 kHz  iq z0  "(no identity)" 109.87.25.148 (LEAVING after 0:00:51)
Fri Oct 16 04:54:04 3d:22:01:14.593 01......  1       TLIMIT exempt local connection from 192.168.2.204
Fri Oct 16 04:54:04 3d:22:01:14.593 01......  1       PWD kiwi SND ALLOWED: config pwd set, but is_local and auto-login set
Fri Oct 16 04:54:24 3d:22:01:34.630 01......  1       "(no identity)" 192.168.2.204 incomplete connection kicked
Fri Oct 16 04:54:47 3d:22:01:58.179 01......  1       TLIMIT exempt local connection from 192.168.2.1
Fri Oct 16 04:54:47 3d:22:01:58.179 01......  1       PWD kiwi SND ALLOWED: config pwd set, but is_local and auto-login set
Fri Oct 16 04:55:14 3d:22:02:24.630 01......  1       "(no identity)" 192.168.2.1 incomplete connection kicked`

I can connect to other public KiwiSDR's just fine, just not locally apparently.  Very strange!

If you have Wireshark, or can read PCAPNG packet files, here is the port 8073 traffic between my SDRAngel client and my local KiwiSDR server.  

https://drive.google.com/file/d/1reGQFB0W8bRwK0HKIDNwZJ9_Y9OItB-U/view?usp=sharing

I compared my session with that of a successful public KiwiSDR session and for some reason my client / server interaction gets hung up after a few WebSocket Ping / WebSocket Pong messages and then quits.

Thanks for your software development efforts.

73

andyz - K1RA

Vort, 2020-10-16:
The difference is here:
[image]

Plugin expects to see `audio_init=0` since this is what I saw in browser.
But in your pcap file `audio_init=1` is located in the same place.

Can you share correct data exchange with local address by browser?
Like on my screenshot (F12 in Chrome) or with Wireshark dump.

azwirko, 2020-10-16:
Here is a PCAPNG file from Wireshark with a few seconds Chrome initially connecting and streaming audio with my KiwiSDR server.

https://drive.google.com/file/d/1E9wr1QdpMullWB6QTJIUfQsxP1pLs7XY/view?usp=sharing

Thanks

Vort, 2020-10-16:
The problem is that I can't see how client reaction to `0` differs from reaction to `1`.
Can you replace `audio_init=0` with `audio_init=1` inside `inputkiwisdr.dll` with hex editor (offset 0xFB23) and try again?
If not, I can send you patched file instead.

azwirko, 2020-10-16:
Vort
  It worked! Here is the PCAPNG for a successful connect and stream of my KiwiSDR on the private LAN IP with SDRAngel and the inputkiwisdr.dll modified with audio_init=1

https://drive.google.com/file/d/10WeCnKsdVQ4BYQLSnRPCcrgSv8GD9-Tp/view?usp=sharing

  I also tried to connect to a number of public SDRs with this modified DLL file and I can not connect.  Here is a sample PCAPNG file of one of those attempts, with the same WebSocket PING / PONG then timeout issue I noted earlier

https://drive.google.com/file/d/1H8r9sCNHUwaIlJ7lmO25pt_DIqR_c_U_/view?usp=sharing

Thanks for your work on debugging this!

Vort, 2020-10-16:
Interesting findings:
[`audio_init` parameter have name `is_local`](https://github.com/jks-prv/Beagle_SDR_GPS/blob/ba95179d1b5f1e1ed880659615a73e6596602d81/web/openwebrx/audio.js#L193), and it perfectly describes what is happening here.
Based on this value, [buffering_scheme is selected](https://github.com/jks-prv/Beagle_SDR_GPS/blob/ba95179d1b5f1e1ed880659615a73e6596602d81/web/openwebrx/audio.js#L285).
Which is not relevant for sdrangel, since plugin is not responsive for buffering here.
So I think that `audio_init` value can be just ignored.

Vort, 2020-10-16:
It is an easy fix. In this condition:
https://github.com/f4exb/sdrangel/blob/8b056708142b0b3b93d6a1f49999775f271372c4/plugins/samplesource/kiwisdr/kiwisdrworker.cpp#L89-L91
`al[1] == "audio_init=0" &&` should be removed. That's all.

But I don't have working build environment right now and I need to free space on my HDD to make build working so I can't test this change right now.
So either I will slowly set my environment, make change, tests and pull request or you and @f4exb can do this change and tests instead of me (theoretically it should be faster).

f4exb, 2020-10-16:
OK that's an easy fix indeed that I can bang in the next release. So let's put back the issue in the queue.

azwirko, 2020-10-16:
Thank you Vort! For now I will just have two inputkiwisdr.dll files, one with audio_init=0 or =1.  I mainly wish to use SDRAngel with my local SDR, but if I wish to connect to a public SDR, then I will copy over the other DLL file until you bang out the next release.  Glad I could help debug this issue with you.

73

f4exb, 2020-10-28:
This change causes a segfault (more on this later).

f4exb, 2020-10-28:
But this works...
```c++
if ((al[1] == "audio_init=0" && al[2] == "audio_rate=12000") || (al[1] == "audio_init=1"))
```

Vort, 2020-10-28:
> This change causes a segfault:

Oops, my mistake, sorry.

This should work:
```if (al.size() > 2 && al[2] == "audio_rate=12000")```

f4exb, 2020-11-03:
Yes this works fine

commit 78b83663 (2020-11-03): KiwiSDR: fixed issue in local network. Fixes #435
commit 35674680 (2020-11-04): KiwiSDR: fixed issue in local network. Fixes #435
closed 2020-11-04 by f4exb