f4exb/sdrangel #437: build error: khadas VIM3 ubuntu bionic arm64

smithrb, 2019-11-02:
clean build throughout following the linux build procedure.

In the final cmake for sdrangel, I'm getting the below.

Any idea?  I'm not having any luck solving this one.

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
SWSCALE_INCLUDE_DIRS (ADVANCED)
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
   used as include directory in directory /opt/build/sdrangel/plugins/channelrx/demoddatv
SWSCALE_LIBRARIES (ADVANCED)
    linked by target "demoddatv" in directory /opt/build/sdrangel/plugins/channelrx/demoddatv

smithrb, 2019-11-04:
Any assist on this?

f4exb, 2020-02-29:
It is pretty obvious that the SWSCALE compoent of ffmpeg is missing. The installation of ffmpeg is incomplete and this issue is to be solved by the user.

However if SWSCALE_INCLUDE_DIRS is not set (NOTFOUND) then FFMPEG_FOUND should not be set and demoddatv build skipped. The FindFFmpeg.cmake file which is barely readable could may be include all the bits needed by demoddatv in the "default" components at: https://github.com/f4exb/sdrangel/blob/master/cmake/Modules/FindFFmpeg.cmake#L36

f4exb, 2020-06-20:
Implemented in v4.14.10 and v5.7.1

closed 2020-06-20 by f4exb