f4exb/sdrangel #439: Problem in modatv compiling from source

justindthomas, 2019-11-06:
I'm using the instructions found here: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux to install SDRAngel on a Pine64+ board running Arch Linux (I intend to use just the server component with a LimeSDR Mini). It mostly compiles okay, but is getting stuck on the modatv plugin:

~~~~~
[ 81%] Building CXX object pluginssrv/channeltx/modatv/CMakeFiles/modatvsrv.dir/atvmod.cpp.o
[ 82%] Built target modfreedvsrv
[ 83%] Built target mimotestmisrv
[ 84%] Built target inputfileinputsrv
Scanning dependencies of target inputtestsourcesrv
[ 84%] Building CXX object pluginssrv/samplesource/testsource/CMakeFiles/inputtestsourcesrv.dir/inputtestsourcesrv_autogen/mocs_compilation.cpp.o
[ 84%] Building CXX object pluginssrv/samplesource/testsource/CMakeFiles/inputtestsourcesrv.dir/testsourceinput.cpp.o
[ 85%] Building CXX object pluginssrv/samplesource/testsource/CMakeFiles/inputtestsourcesrv.dir/testsourceplugin.cpp.o
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'void ATVMod::pullVideo(Real&)':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:349:66: error: 'CV_BGR2GRAY' was not declared in this scope
                   cv::cvtColor(colorFrame, m_videoframeOriginal, CV_BGR2GRAY);
                                                                  ^~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:469:75: error: 'CV_BGR2GRAY' was not declared in this scope
                     cv::cvtColor(colorFrame, camera.m_videoframeOriginal, CV_BGR2GRAY);
                                                                           ^~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'virtual bool ATVMod::handleMessage(const Message&)':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:579:39: error: 'CV_CAP_PROP_POS_FRAMES' was not declared in this scope
             framesCount = m_video.get(CV_CAP_PROP_POS_FRAMES);;
                                       ^~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'void ATVMod::openImage(const QString&)':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:810:56: error: 'CV_LOAD_IMAGE_GRAYSCALE' was not declared in this scope
     m_imageFromFile = cv::imread(qPrintable(fileName), CV_LOAD_IMAGE_GRAYSCALE);
                                                        ^~~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'void ATVMod::openVideo(const QString&)':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:840:34: error: 'CV_CAP_PROP_FPS' was not declared in this scope
         m_videoFPS = m_video.get(CV_CAP_PROP_FPS);
                                  ^~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:840:34: note: suggested alternative: 'CV_PROP_RW'
         m_videoFPS = m_video.get(CV_CAP_PROP_FPS);
                                  ^~~~~~~~~~~~~~~
                                  CV_PROP_RW
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:841:42: error: 'CV_CAP_PROP_FRAME_WIDTH' was not declared in this scope
         m_videoWidth = (int) m_video.get(CV_CAP_PROP_FRAME_WIDTH);
                                          ^~~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:842:43: error: 'CV_CAP_PROP_FRAME_HEIGHT' was not declared in this scope
         m_videoHeight = (int) m_video.get(CV_CAP_PROP_FRAME_HEIGHT);
                                           ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:843:43: error: 'CV_CAP_PROP_FRAME_COUNT' was not declared in this scope
         m_videoLength = (int) m_video.get(CV_CAP_PROP_FRAME_COUNT);
                                           ^~~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:844:47: error: 'CV_CAP_PROP_FOURCC' was not declared in this scope
         int ex = static_cast<int>(m_video.get(CV_CAP_PROP_FOURCC));
                                               ^~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:844:47: note: suggested alternative: 'CV_PROP_RW'
         int ex = static_cast<int>(m_video.get(CV_CAP_PROP_FOURCC));
                                               ^~~~~~~~~~~~~~~~~~
                                               CV_PROP_RW
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'void ATVMod::seekVideoFileStream(int)':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:939:21: error: 'CV_CAP_PROP_POS_FRAMES' was not declared in this scope
         m_video.set(CV_CAP_PROP_POS_FRAMES, seekPoint);
                     ^~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'void ATVMod::scanCameras()':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:957:64: error: 'CV_CAP_PROP_FPS' was not declared in this scope
    m_cameras.back().m_videoFPS = m_cameras.back().m_camera.get(CV_CAP_PROP_FPS);
                                                                ^~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:957:64: note: suggested alternative: 'CV_PROP_RW'
    m_cameras.back().m_videoFPS = m_cameras.back().m_camera.get(CV_CAP_PROP_FPS);
                                                                ^~~~~~~~~~~~~~~
                                                                CV_PROP_RW
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:958:72: error: 'CV_CAP_PROP_FRAME_WIDTH' was not declared in this scope
    m_cameras.back().m_videoWidth = (int) m_cameras.back().m_camera.get(CV_CAP_PROP_FRAME_WIDTH);
                                                                        ^~~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:959:73: error: 'CV_CAP_PROP_FRAME_HEIGHT' was not declared in this scope
    m_cameras.back().m_videoHeight = (int) m_cameras.back().m_camera.get(CV_CAP_PROP_FRAME_HEIGHT);
                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp: In member function 'void ATVMod::mixImageAndText(cv::Mat&)':
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:1030:153: error: 'CV_AA' was not declared in this scope
     cv::putText(image, m_settings.m_overlayText.toStdString(), textOrg, fontFace, fontScale, cv::Scalar::all(255*m_settings.m_uniformLevel), thickness, CV_AA);
                                                                                                                                                         ^~~~~
/opt/build/sdrangel/plugins/channeltx/modatv/atvmod.cpp:1030:153: note: suggested alternative: 'CV_PI'
     cv::putText(image, m_settings.m_overlayText.toStdString(), textOrg, fontFace, fontScale, cv::Scalar::all(255*m_settings.m_uniformLevel), thickness, CV_AA);
                                                                                                                                                         ^~~~~
                                                                                                                                                         CV_PI
make[2]: *** [pluginssrv/channeltx/modatv/CMakeFiles/modatvsrv.dir/build.make:76: pluginssrv/channeltx/modatv/CMakeFiles/modatvsrv.dir/atvmod.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2901: pluginssrv/channeltx/modatv/CMakeFiles/modatvsrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Building CXX object pluginssrv/samplesource/testsource/CMakeFiles/inputtestsourcesrv.dir/testsourcethread.cpp.o
[ 85%] Linking CXX shared library ../../lib/libsoapysdrdevice.so
[ 85%] Built target soapysdrdevice
[ 85%] Building CXX object pluginssrv/samplesource/testsource/CMakeFiles/inputtestsourcesrv.dir/testsourcesettings.cpp.o
[ 85%] Building CXX object pluginssrv/samplesource/testsource/CMakeFiles/inputtestsourcesrv.dir/testsourcewebapiadapter.cpp.o
[ 85%] Linking CXX shared library ../../../lib/pluginssrv/libinputtestsourcesrv.so
[ 85%] Built target inputtestsourcesrv
make: *** [Makefile:152: all] Error 2
~~~~~

This is the cmake command I'm using:

~~~~~
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
    -DLIMESUITE_DIR=/opt/install/LimeSuite \
    -DSOAPYSDR_DIR=/opt/install/SoapySDR \
    -DCM256CC_DIR=/opt/install/cm256cc \
    -DDSDCC_DIR=/opt/install/dsdcc \
    -DSERIALDV_DIR=/opt/install/serialdv \
    -DMBE_DIR=/opt/install/mbelib \
    -DCODEC2_DIR=/opt/install/codec2 \
    -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
~~~~~

I was initially building with `-DBUILD_GUI=OFF` and without SoapySDR when I first hit the error, but have sort of been flailing, adding things in to try to get it to continue. I'm not having any luck.

Any advice would be greatly appreciated.

justindthomas, 2019-11-06:
Okay, I got my hands a little dirty (it's been a long time since I've thought much in C++) and tracked down the updated constant names. Changing things in this way gets it to compile:

~~~~~
diff --git a/plugins/channeltx/modatv/atvmod.cpp b/plugins/channeltx/modatv/atvmod.cpp
index 8892fcc21..b138dbfc6 100644
--- a/plugins/channeltx/modatv/atvmod.cpp
+++ b/plugins/channeltx/modatv/atvmod.cpp
@@ -346,7 +346,7 @@ void ATVMod::pullVideo(Real& sample)
             		        mixImageAndText(colorFrame);
             		    }
 
-            		    cv::cvtColor(colorFrame, m_videoframeOriginal, CV_BGR2GRAY);
+            		    cv::cvtColor(colorFrame, m_videoframeOriginal, cv::COLOR_BGR2GRAY);
             		    resizeVideo();
             		}
             	}
@@ -466,7 +466,7 @@ void ATVMod::pullVideo(Real& sample)
                         mixImageAndText(colorFrame);
                     }
 
-                    cv::cvtColor(colorFrame, camera.m_videoframeOriginal, CV_BGR2GRAY);
+                    cv::cvtColor(colorFrame, camera.m_videoframeOriginal, cv::COLOR_BGR2GRAY);
                     resizeCamera();
                 }
 
@@ -576,7 +576,7 @@ bool ATVMod::handleMessage(const Message& cmd)
 
         if (m_videoOK && m_video.isOpened())
         {
-            framesCount = m_video.get(CV_CAP_PROP_POS_FRAMES);;
+            framesCount = m_video.get(cv::CAP_PROP_POS_FRAMES);;
         } else {
             framesCount = 0;
         }
@@ -807,7 +807,7 @@ void ATVMod::applyStandard()
 
 void ATVMod::openImage(const QString& fileName)
 {
-    m_imageFromFile = cv::imread(qPrintable(fileName), CV_LOAD_IMAGE_GRAYSCALE);
+    m_imageFromFile = cv::imread(qPrintable(fileName), cv::ImreadModes::IMREAD_GRAYSCALE);
 	m_imageOK = m_imageFromFile.data != 0;
 
 	if (m_imageOK)
@@ -837,11 +837,11 @@ void ATVMod::openVideo(const QString& fileName)
     if (m_videoOK)
     {
         m_settings.m_videoFileName = fileName;
-        m_videoFPS = m_video.get(CV_CAP_PROP_FPS);
-        m_videoWidth = (int) m_video.get(CV_CAP_PROP_FRAME_WIDTH);
-        m_videoHeight = (int) m_video.get(CV_CAP_PROP_FRAME_HEIGHT);
-        m_videoLength = (int) m_video.get(CV_CAP_PROP_FRAME_COUNT);
-        int ex = static_cast<int>(m_video.get(CV_CAP_PROP_FOURCC));
+        m_videoFPS = m_video.get(cv::CAP_PROP_FPS);
+        m_videoWidth = (int) m_video.get(cv::CAP_PROP_FRAME_WIDTH);
+        m_videoHeight = (int) m_video.get(cv::CAP_PROP_FRAME_HEIGHT);
+        m_videoLength = (int) m_video.get(cv::CAP_PROP_FRAME_COUNT);
+        int ex = static_cast<int>(m_video.get(cv::CAP_PROP_FOURCC));
         char ext[] = {(char)(ex & 0XFF),(char)((ex & 0XFF00) >> 8),(char)((ex & 0XFF0000) >> 16),(char)((ex & 0XFF000000) >> 24),0};
 
         qDebug("ATVMod::openVideo: %s FPS: %f size: %d x %d #frames: %d codec: %s",
@@ -936,7 +936,7 @@ void ATVMod::seekVideoFileStream(int seekPercentage)
     if ((m_videoOK) && m_video.isOpened())
     {
         int seekPoint = ((m_videoLength * seekPercentage) / 100);
-        m_video.set(CV_CAP_PROP_POS_FRAMES, seekPoint);
+        m_video.set(cv::CAP_PROP_POS_FRAMES, seekPoint);
         m_videoFPSCount = m_videoFPSq;
         m_videoPrevFPSCount = 0;
         m_videoEOF = false;
@@ -954,9 +954,9 @@ void ATVMod::scanCameras()
 
 		if (m_cameras.back().m_camera.isOpened())
 		{
-			m_cameras.back().m_videoFPS = m_cameras.back().m_camera.get(CV_CAP_PROP_FPS);
-			m_cameras.back().m_videoWidth = (int) m_cameras.back().m_camera.get(CV_CAP_PROP_FRAME_WIDTH);
-			m_cameras.back().m_videoHeight = (int) m_cameras.back().m_camera.get(CV_CAP_PROP_FRAME_HEIGHT);
+			m_cameras.back().m_videoFPS = m_cameras.back().m_camera.get(cv::CAP_PROP_FPS);
+			m_cameras.back().m_videoWidth = (int) m_cameras.back().m_camera.get(cv::CAP_PROP_FRAME_WIDTH);
+			m_cameras.back().m_videoHeight = (int) m_cameras.back().m_camera.get(cv::CAP_PROP_FRAME_HEIGHT);
 
 			//m_cameras.back().m_videoFPS = m_cameras.back().m_videoFPS < 0 ? 16.3f : m_cameras.back().m_videoFPS;
 
@@ -1027,7 +1027,7 @@ void ATVMod::mixImageAndText(cv::Mat& image)
     // position the text in the top left corner
     cv::Point textOrg(6, textSize.height+10);
     // then put the text itself
-    cv::putText(image, m_settings.m_overlayText.toStdString(), textOrg, fontFace, fontScale, cv::Scalar::all(255*m_settings.m_uniformLevel), thickness, CV_AA);
+    cv::putText(image, m_settings.m_overlayText.toStdString(), textOrg, fontFace, fontScale, cv::Scalar::all(255*m_settings.m_uniformLevel), thickness, cv::LINE_AA);
 }
 
 void ATVMod::applyChannelSettings(int outputSampleRate, int inputFrequencyOffset, bool force)
~~~~~

f4exb, 2019-11-06:
Most likely you are not using an OpenCV version that is compatible with SDRangel code. Please stick to the given versions. Moreover if you are running on ARM64 in server mode you are better with building a Docker image: see: https://github.com/f4exb/sdrangel-docker

There are also artifacts in the Docker Hub repository: https://hub.docker.com/r/f4exb06/sdrangelsrv
then `docker pull f4exb06/sdrangelsrv` is all you need.

justindthomas, 2019-11-06:
Thanks. The comment about OpenCV is certainly correct. But it's not something I installed deliberately - it's whatever was in the packages available from Arch. The instructions seem to go to some length to avoid the problem of mismatched versions by calling out components that need to be compiled. Maybe OpenCV should be added?

I use Docker a lot, and I can see the appeal of using it to crystallize installation variables. I'll be honest, I didn't even know you could do USB passthrough with Docker. (I'm using a LimeSDR Mini.)

Regardless, that's a lot of overhead to require for a SBC like the Pine A64+ I'm installing on.

I'll close this ticket as there's nothing substantive to be done. Hopefully it can serve as a resource when folks encounter this error in the future.

closed 2019-11-06 by justindthomas