f4exb/sdrangel #441: HackRF - Windows AND MacOS ERROR: "could not start sample source"

aneitzel, 2019-11-11:
Plattform OS: Win10 latest / same as on MacOS Mojave and HighSierra
Affected: SDRangel allmost all versions

Note: HackRF is working with all other Tools without issues...  on my Win10 and MacOS devices, which is important. So HackRF is ok and working fine.

Problem: Installed SDRangel and I always got error "could not start sample source" after klick on R0 "play" to start sampling. Seems there is something wrong in handling of SDRangel and HackRF. I tested it on Windows with https://github.com/f4exb/sdrangel/releases/download/v4.11.12/sdrangel-4.11.12-win64.exe and also on MacOS with ready DMG image AND full compiled via macports.

Any other users which has issues with HackRF and SDRangel here? Any hints?

Thanks & regards André

aneitzel, 2019-11-11:
<blockquote class="imgur-embed-pub" lang="en" data-id="a/YoJUqC0"><a href="//imgur.com/a/YoJUqC0"></a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

aneitzel, 2019-11-12:
**Here some additional Infos gathered on MacOS:**

Error seen in Terminal during start of SDRangel:

[INFO] [UHD] Mac OS; Clang version 10.0.1 (clang-1001.0.46.4); Boost_107100; UHD_3.14.1.1-MacPorts-Release
get_devices started
Device count: 0
get_devices end
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.13.5
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy soapy 
Using HackRF One with firmware git-eae870d 
...
[INFO] Opening HackRF One #0 14d463dc2f391ee1...
[INFO] Could not Open HackRF Device
2019-11-12 19:09:18.370 (W) DeviceSoapySDRScan::scan: hackrf #0 cannot be opened: hackrf open failed
gf256_ctx::gf256_init_: initialized
...

**So i checked SoapySDR with find option:**

Andres-iMac:~ alpha$ SoapySDRUtil -find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 14d463dc2f391ee1
  part_id = a000cb3c00784758
  serial = 000000000000000014d463dc2f391ee1
  version = git-eae870d

Pse note that all other SDR tools has no issue to talk to HackRF, i have no idea whats wrong :-(

f4exb, 2019-11-15:
Why use SoapySDR when HackRF is natively supported?

aneitzel, 2019-11-15:
I dont configured anything with soapySDR - the output is what i see after run SDRangel. My HackRF itself will be seen in SDRangel, but i always got ERROR: "could not start sample source" then i try to start sampling.  Later i found the message related to soapy i guessed its maybe a issue with it. 

But if HackRF is natively supported, is it possible that your native support pnly work with a fix fw level? Or native support + load of soapy module at same time makes any problems? 

Let me make clear, that my hackRF is working properly with all other SDR tools, under Win10 and also MacOS. This is the output from HackRF_info:

d:\HackRF\HackRF_tools>hackrf_info.exe
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Board ID Number: 2 (HackRF One)
Firmware Version: git-eae870d (API:1.03)
Part ID Number: 0xa000cb3c 0x00784758

Thanks for help.

aneitzel, 2019-11-15:
Found the problem: !!!!

Your native support only support mossmanns build. If someone use his own compiled firmware from "git" or HackRF Portapack (in native HackRF mode) SDRangel fails!

Here is the output from hackrf_info with 2018.01.1 (API:1.02) fw flashed:

d:\HackRF\HackRF_tools>hackrf_info.exe
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f391ee1
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00784758

This firmwarepackage is working and SDR angel can access HackRF

If i flash back to newer Firmware Version: git-eae870d (API:1.03) SDRangel fails! All other SDR tools not :-(

f4exb, 2019-11-16:
Thanks for unveiling the issue with Portapack. Of course only Mossman's (author of HackRF) version is supported and will be supported. If the Portapack offspring did not stay compatible it is the problem of Portapack. So sorry but SDRangel will stay with the latest stable release of original HackRF software and firmware that is 2018.01.1

This is the same issue as described in #392

closed 2019-12-19 by f4exb