f4exb/sdrangel #445: crashing while using 2 pluto

patel999jay, 2019-11-20:
SDRangel is unexpectedly crashing down when i connect two different pluto devices with same.

Radiomix2000, 2019-12-09:
I have the same issue. Only once I was lucky to open the list of available SDRs, and there was no Pluto SDRs available. It means both of them were not identified by SDRangel. Maybe it's enumeration issues, don't know

f4exb, 2019-12-14:
So this is with Pluto on Ethernet interface. This would have been good to know since the beginning. Support for networked Pliutos is crap anyway and needs rework. It scans the network and spends ~5s even if there is no Pluto connected so it slows down the start sequence of SDRangel for everybody. A feature to use device related arguments was introduced some time ago for SoapySDR devices. This could be extended to Pluto in order to specify their address for networked Plutos.

patel999jay, 2019-12-14:
Hi ,

Sorry about that. I was new for this field. I am trying to connect 2 Pluto with sdrangle, when i start sdrangle it recognized both sdr itself, what i am trying to do is use 1 tx -sdr1 and 1 rx - sdr2 (same frequency)  and 1 tx -sdr2 and 1 rx - sdr1 (same frequency). so basically use alternate channels from pluto to send signals. but when you select 2nd pluto it was unexpectedly crashed sdrangle for some reason.

f4exb, 2019-12-14:
OK so the trouble starts when there are 2 Pluto that can be discovered in the network. I cannot try this configuration myself but my guess is that the problem is precisely the auto discovery because it cannot handle many devices in the same network. In fact it uses `ip:pluto.local` for the URI which does not mention anything about the actual IP address.

More generally the idea would be to allow the user to declare not auto discoverable devices similarly to what exists in SDR console.

f4exb, 2019-12-17:
This has been delivered in v4.12.3 you now have to specify the address of the pluto as an `uri=ip:x.x.x.x` parameter as mentioned in the release notes and documentation. If you have more than one Pluto you need to create a "non discoverable" device for each one with their respective address.

closed 2019-12-17 by f4exb