f4exb/sdrangel #446: Saving a bookmark does not work if only the modulation changes

febs, 2019-11-22:
Linux, git commit acd08925368ba1ee2f35469314abdc4dea02a1fe (v 4.12.1).

How I can reproduce it.
Create a bookmark.
Then add a demodulation plug-in.
Press save.
Reload the bookmark and it was as before saving the changes.

Thanks,

f4exb, 2019-12-20:
You do not understand how this works although the term "bookmark" does give an idea because it behaves a little bit like browser bookmarks. However this is called "presets". So when you "save" presets you save the whole bunch of presets to disk including the current configuration on R0. So "save" is irrelevant in your scenario, This is what actually happens:
- "create a bookmark": this creates a preset without demodulator (your current state)
- "add demodulation plugin": this adds a demod to your device set (nothing has changed for the preset)
- "save": this saves all presets including the empty one (no demod) you just created
- "reload the bookmark": the preset is still empty and this is what you see

What you are looking for is "update preset" (the cycle arrow - second icon from the left) after you have added the demodulator.

febs, 2019-12-21:
Thanks for the explaination.

closed 2019-12-21 by febs