f4exb/sdrangel #449: Entry point not found

g4bao, 2019-11-26:
Very new to Github. Hope I'm doing this right! 

[image]

I'm getting the above error on startup, Clicking OK seems to run the program OK

Vort, 2019-12-28:
It may happen because non-compatible library is found in your system PATH variable.
You can try temporary rename that PothosSDR folder and check if error will happen again.

g4bao, 2019-12-28:
I renamed the PothosSDR folder as you suggested and the error does not occur. When returning the folder to its original name., the error message reoccurs. As the folder was created by the Pothos Windows installer, I'm not sure what the next step is.
John

Vort, 2019-12-28:
Do you need that PATH modifications for other software?
Looks like they can be disabled while installing PothosSDR:
https://i.imgur.com/pC9uKY6.png

g4bao, 2019-12-28:
Perfect! I removed the Path to PothosSDR/bin and the error does not now occur. Pothos continues to work correctly! 
Thanks for the help

closed 2019-12-28 by g4bao