f4exb/sdrangel #451: [Test Report] Works on OpenPOWER!

madscientist159, 2019-12-05:
If you wanted to add a note to the main page, I've tested this on an OpenPOWER (ppc64le, an open ISA like RISC-V but with far more powerful CPUs available now) POWER9 sytem with an RTL2838 USB stick.  It works very well, CPU load is minimal, I get the impression some fairly advanced filters could be used on these machines with ease.

f4exb, 2020-01-16:
There seems to be a lot of POWER9 systems can you please detail a bit more your system (board name, commercial availability, etc...)?

madscientist159, 2020-01-16:
Sure, this was tested on a Talos II from Raptor Computing Systems -- it's a COTS workstation / PC with several variants, e.g. https://raptorcs.com/content/TLSDS3/intro.html

It should also work on the Blackbird systems (same vendor), but I haven't had a chance to test that yet.  For the most part if it works on one POWER9 system, it's going to work on all of them, the OpenPOWER ecosystem is very similar to the x86 ecosystem in that respect.

f4exb, 2020-01-17:
Interesting. Not an entry level system but for sure very capable and I like the Open Power concept. Does it run on the vendor provided Debian Buster?

f4exb, 2020-01-17:
I think I will move the hardware info paragraph to the Wiki to be more flexible.

madscientist159, 2020-01-17:
> Interesting. Not an entry level system but for sure very capable and I like the Open Power concept. Does it run on the vendor provided Debian Buster?

Yes it does.  Essentially all of the major distros have offerings for OpenPOWER (e.g. Ubuntu, RHEL, CentOS, Fedora, OpenSUSE, etc.), but I only tested sdrangel on Debian Buster.

f4exb, 2020-05-20:
Done

closed 2020-05-20 by f4exb