f4exb/sdrangel #454: Error while compiling OpenCV constants undeclared in this scope.

Dantali0n, 2019-12-13:
When compiling sdrangel from master I get the following errors:

"‘CV_BGR2GRAY’ was not declared in this scope" it seems the atv related plugins are dependend on a old version of OpenCV. I am on an Arch based system which has OpenCV 4.* not 2.4 or 3.2.

I understand that updating sdrangel to work with a recent version of OpenCV might be a time consuming endeavor. I was looking for a way to disable plugins upon compilation but coulnd't find the relevant CMake flags in the documentation.

Machine details:
```
inxi -v3
System:    Host: Dantali0n Kernel: 5.4.2-1-MANJARO x86_64 bits: 64 compiler: gcc v: 9.2.0 Desktop: i3 4.17.1 
           Distro: Manjaro Linux 
Machine:   Type: Laptop System: Notebook product: P64_HJ,HK1 v: N/A serial: <root required> 
           Mobo: Notebook model: P64_HJ,HK1 serial: <root required> UEFI: American Megatrends v: 1.05.08 date: 07/06/2017 
Battery:   ID-1: BAT0 charge: 41.8 Wh condition: 41.8/45.2 Wh (93%) model: Notebook BAT status: Full 
CPU:       Topology: Quad Core model: Intel Core i7-7700HQ bits: 64 type: MT MCP arch: Kaby Lake rev: 9 L2 cache: 6144 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 44817 
           Speed: 2801 MHz min/max: 800/2800 MHz Core speeds (MHz): 1: 2801 2: 2801 3: 2800 4: 2802 5: 2800 6: 2800 7: 2801 
           8: 2802 
Graphics:  Device-1: Intel HD Graphics 630 vendor: CLEVO/KAPOK driver: i915 v: kernel bus ID: 00:02.0 
           Device-2: NVIDIA GP107M [GeForce GTX 1050 Ti Mobile] vendor: CLEVO/KAPOK driver: nvidia v: 430.64 bus ID: 01:00.0 
           Display: x11 server: X.Org 1.20.6 driver: intel unloaded: nvidia resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa DRI Intel HD Graphics 630 (Kaby Lake GT2) v: 4.5 Mesa 19.2.7 direct render: Yes 
Network:   Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: CLEVO/KAPOK driver: r8168 v: 8.047.05-NAPI 
           port: d000 bus ID: 03:00.1 
           IF: enp3s0f1 state: up speed: 1000 Mbps duplex: full mac: 80:fa:5b:49:b1:1b 
           Device-2: Intel Wireless-AC 9260 vendor: Bigfoot Networks driver: iwlwifi v: kernel port: d000 bus ID: 04:00.0 
           IF: wlp4s0 state: down mac: 42:46:c0:48:2f:2b 
           IF-ID-1: docker0 state: up speed: N/A duplex: N/A mac: 02:42:3b:47:b0:55 
           IF-ID-2: veth2142ca7 state: up speed: 10000 Mbps duplex: full mac: 22:f7:78:88:93:8c 
Drives:    Local Storage: total: 1.94 TiB used: 835.80 GiB (42.2%) 
Info:      Processes: 365 Uptime: 1h 20m Memory: 31.25 GiB used: 8.09 GiB (25.9%) Init: systemd Compilers: gcc: 9.2.0 
           clang: 9.0.0 Shell: bash v: 5.0.11 inxi: 3.0.37
```

Dantali0n, 2019-12-13:
I think it would be quite a minimal patch to have the modatv work with both OpenCV > 3.2 and below as it seems that all these constant have just become part of the cv namespace.

So `CV_BGR2GRAY` becomes `cv:COLOR_BGR2GRAY`.  See https://github.com/Dovyski/cvui/issues/52

A minimal wrapper with some `#ifdef` could allow to support the entire range of versions.

f4exb, 2019-12-14:
SDRangel is based on Ubuntu 18.04 and OpenCV 3.2 is the version that is used there. I have no plans to change this not make any patch to use a different version of OpenCV. Other Linux distributions can use Docker to build and run an image based on Ubuntu 18.04

closed 2019-12-14 by f4exb