f4exb/sdrangel #455: Symbol lookup error on Buster

sv1, 2019-12-13:
I am trying to run sdrangel 4.12.1 on Debian buster.

On startup I get this error:
`
...........

Found Rafael Micro R820T tuner
2019-12-13 22:16:55.781 (I) RTLSDRInput::openDevice: open: Realtek RTL2838UHIDIR, SN: 00000001
2019-12-13 22:16:55.857 (D) RTLSDRInput::openDevice:  29 gains
2019-12-13 22:16:55.860 (D) DSPDeviceSourceEngine::addSink:  FileRecord
2019-12-13 22:16:55.860 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2019-12-13 22:16:55.860 (D) FileRecord::handleMessage: DSPSignalNotification: m_inputSampleRate:  0  m_centerFrequency:  0
2019-12-13 22:16:55.885 (D) DSPDeviceSourceEngine::setSource
2019-12-13 22:16:55.885 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPSetSource
2019-12-13 22:16:55.885 (D) DSPDeviceSourceEngine::gotoIdle
2019-12-13 22:16:55.885 (D) DSPDeviceSourceEngine::handleSetSource: set RTL2838UHIDIR (SN 00000001)
2019-12-13 22:16:55.892 (D) RTLSDRGui::updateFrequencyLimits: delta: 0 min: 24000 max: 1900000
2019-12-13 22:16:55.892 (D) RTLSDRGui::updateFrequencyLimits: delta: 0 min: 24000 max: 1900000
2019-12-13 22:16:55.897 (D) RTLSDRInput::applySettings: AGC mode off
2019-12-13 22:16:55.897 (D) DSPDeviceSourceEngine::configureCorrections
2019-12-13 22:16:55.897 (D) RTLSDRInput::applySettings: corrections: DC block: false IQ imbalance: false
2019-12-13 22:16:55.897 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPConfigureCorrection
2019-12-13 22:16:55.897 (C) RTLSDRInput::applySettings: could not set LO ppm correction: 0
2019-12-13 22:16:55.943 (D) RTLSDRInput::applySettings: sample rate set to 1024000
2019-12-13 22:16:55.943 (D) RTLSDRInput::applySettings: log2decim set to 4
2019-12-13 22:16:55.943 (D) RTLSDRInput::applySettings: set fc pos (enum) to  2
2019-12-13 22:16:55.943 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  435000000  Hz  device center freq:  435000000  Hz  device sample rate:  1024000 S/s  Actual sample rate:  64000 S/s  center freq position code:  2  image frequency:  435000000 Hz
2019-12-13 22:16:55.972 (D) RTLSDRInput::applySettings: rtlsdr_set_center_freq(435000000)
2019-12-13 22:16:55.972 (D) RTLSDRInput::applySettings: set noModMode to  false
Disabled direct sampling mode
sdrangel: symbol lookup error: /usr/lib/sdrangel/plugins/libinputrtlsdr.so: undefined symbol: rtlsdr_set_tuner_bandwidth
`
Any help would be greatly appreciated!

f4exb, 2019-12-14:
librtlsdr is a package dependency on Linux builds maybe this is not such a good idea (in Windows this is compiled from source). librtlsdr is easy to build from source (unlike OpenCV for example) so it could be worth to move it to the "external" build section, Moreover a different version only available from source could be used in the future to support Kerberos SDR. In the meantime distribution of Buster version is suspended.

f4exb, 2019-12-17:
Delivered in v4.12.4

closed 2019-12-17 by f4exb