f4exb/sdrangel #456: RTL TCP

gaspar1987, 2019-12-15:
Hello, I'd like to know how we use your tcp.
To activate the software.
Thank you for giving me more information on this to help me solve the problem. Thank you.
[image]

f4exb, 2019-12-15:
You cannot use RTL-SDR via TCP it has to be connected to your system via USB. There are no plans to have RTL-TCP supported... period.

closed 2019-12-15 by f4exb

dterweij, 2020-07-28:
Oh that is pitty. My RTL_SDR is plugged in at the PC located in the attic. Close to the antenna (3 meters of cable). 90% of all tools I do use is over RTL_TCP connection to my PC downstairs.  Please try to plan it...  I am not planning to move the equipment from attic to downstairs. Period :)

f4exb, 2020-07-28:
This issue was closed for two good reasons:
  - Streaming real time data at relatively fast rate over TCP is a bad idea so I will not promote it
  - SDRangel has its own options for remote operation that use UDP and FEC redundancy for protection against block loss. Please check these plugins:
    - https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/remotesink/readme.md (sender side)
    - https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/remoteinput/readme.md (receiver side)

In addition there is a server flavor of SDRangel that lets you run it on a RPi4 (for example) as a server. There are Docker images in Docker hub to just do that: https://hub.docker.com/r/f4exb06/sdrangelsrv The sdrangel-docker ancillary project: https://github.com/f4exb/sdrangel-docker contains more info about running SDRangel in Docker.

So really no reason to support RTL TCP.

synthead, 2021-09-21:
I would really like to see rtl_tcp support, even if there are other options :+1:

betzburger, 2022-05-26:
It looks like sdrangel is a really nice program with many features and I would love to try it. My Raspberry 4 is in the attic and the only way to use it by now is to send the data with rtl_tcp to the network and take the data back from the network with gqrx. This is how I do at the moment. For me (sorry) it seems it is super complicated to do the same with sdrangel / sdrangelsrv. At least I can't find any easy to follow step-by-step instruction. If there is some it would be fantastic.

icedream, 2022-06-18:
The problem I currently have with this is I use a range of tools to process data via the network, all using RTL-TCP. SDRangel is one of the very few apps that decidedly do not cooperate with this protocol and would require exclusivity over the device to set up a server, stopping the functionality of all other apps the moment I would migrate to its solution.

If there were a way to at least set up a server that can take RTL-TCP via localhost and convert it to something SDRangel can work with more efficiently, I'd have less trouble using it. But as of right now this effectively means I can't use SDRangel.

f4exb, 2022-06-18:
I have zero motivation to write a plugin for rtl-tcp but of course you are welcome to contribute

c3kkos, 2023-10-10:
Hi there, i agree with you, f4exb, regarding the rtl_tcp option. too high bandwidth is required to transfer raw data.

SO my aim is to use your remote sink but i don't know how to use it. Must i compile it from a git clone?

srcejon, 2023-10-10:
Whether you use UDP or TCP isn't really going to change the bandwidth requirement.

To reduce the bandwidth, lower sample rate as much as possible and make sure you're using 8-bit transfers.

I have been messing around with a few ways of supporting compressed IQ streams in Remote TCP Sink and Input plugins, but, not ready yet.

c3kkos, 2023-10-11:
The only srv i know that has data compression is spyserver. Closed source. YET it works with sdr++ but with severe limitations in bandwidth