f4exb/sdrangel #457: getting text output from LoRa Demodulator?

Dieter9, 2019-12-16:
I am testing the LoRa-Decoder of SDRangel but do not see the decoded Text.
I am running an aktuell version of SDRangel from git (master) on a Linux system.
In an older version it was possible to see the decoded text on stdout, but now I am not able to find any way to see the text or data output.

I hope, some one could give me a hint.

f4exb, 2019-12-17:
Lora demodulator is not supposed to work in fact. It has only gone through global redesigns but needs a complete review and rework. This is followed up on this ticket: #309

f4exb, 2020-02-26:
Implemented in v5.2.0

closed 2020-02-26 by f4exb

Dieter9, 2020-02-27:
Thanks, that's good news. Unfortunately I can't compile SDRangel on my Arch-Linux system since a few weeks. I got the following Error message:
```
[ 52%] Building CXX object pluginssrv/channeltx/modfreedv/CMakeFiles/modfreedvsrv.dir/freedvmod.cpp.o 
In file included from /usr/include/qt/QtCore/qnamespace.h:43,
                 from /usr/include/qt/QtCore/qobjectdefs.h:48,
                 from /usr/include/qt/QtCore/qobject.h:46,
                 from /usr/include/qt/QtCore/QObject:1,
                 from /home/my/SDRangel/plugins/channeltx/remotesource/remotesource.h:21,
                 from /home/my/SDRangel/build/pluginssrv/channeltx/remotesource/remotesourcesrv_autogen/EWIEGA46WW/moc_remotesource.cpp:10,
                 from /home/my/SDRangel/build/pluginssrv/channeltx/remotesource/remotesourcesrv_autogen/mocs_compilation.cpp:2:
/usr/include/qt/QtCore/qglobal.h: In instantiation of 'T qExchange(T&, U&&) [with T = QDebug::Stream*; U = long int]':
/usr/include/qt/QtCore/qdebug.h:118:77:   required from here
/usr/include/qt/QtCore/qglobal.h:1050:7: error: invalid conversion from 'long int' to 'QDebug::Stream*' [-fpermissive]
 1050 |     t = std::forward<U>(newValue);
      |     ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       |
      |       long int
[ 52%] Building CXX object pluginssrv/channeltx/modatv/CMakeFiles/modatvsrv.dir/atvmod.cpp.o
make[2]: *** [pluginssrv/channeltx/remotesource/CMakeFiles/remotesourcesrv.dir/build.make:63: pluginssrv/channeltx/remotesource/CMakeFiles/remotesourcesrv.dir/remotesourcesrv_autogen/mocs_compilation.cpp.o] Error 1
make[2]: Leaving directory '/home/my/SDRangel/build'
make[1]: *** [CMakeFiles/Makefile2:7471: pluginssrv/channeltx/remotesource/CMakeFiles/remotesourcesrv.dir/all] Error 2
```
The error occurs since the update of QT5 to version 5.14. I can't decide whether this indicates an error in SDRangel or a bug in QT5.

f4exb, 2020-02-27:
Hi,

this is an issue with Qt 5.14 and the compiler according to this issue: https://gitlab.com/pholy/OSCAR-code/issues/5

I think you have to add -fpermissive flag to the compiler option. As this is Arch specific I will not add it upstrream so you will have to update it on your side.

Brgds.

flaviut, 2020-05-07:
I really don't understand what's going on here, but if anyone's looking to patch this more minimally, the fix would be to apply the following onto `qt/QtCore/qdebug.h`:

```diff
     inline QDebug(QString *string) : stream(new Stream(string)) {}
     inline QDebug(QtMsgType t) : stream(new Stream(t)) {}
     inline QDebug(const QDebug &o):stream(o.stream) { ++stream->ref; }
-    QDebug(QDebug &&other) noexcept : stream{qExchange(other.stream, nullptr)} {}
+    QDebug(QDebug &&other) noexcept : stream{qExchange(other.stream, static_cast<Stream*>(nullptr))} {}
     inline QDebug &operator=(const QDebug &other);
     QDebug &operator=(QDebug &&other) noexcept
     { QDebug{std::move(other)}.swap(*this); return *this; }
```