f4exb/sdrangel #460: Advise pls, crash on soapy remote device selection

9a4db, 2019-12-29:
Ubuntu 18.04 story, clean installation,
SDRAngel installed like advised Unix installation:

q@q-X756UX:/opt/build/sdrangel/build$ ls /usr/lib/x86_64-linux-gnu/libSoapy* /usr/local/lib/libSoapy* /usr/include/Soapy* /usr/local/include/Soapy*
/usr/lib/x86_64-linux-gnu/libSoapySDR.so /usr/local/lib/libSoapySDR.so.0.8
/usr/local/lib/libSoapySDR.so /usr/local/lib/libSoapySDR.so.0.8.0

/usr/include/SoapySDR:
Config.h Device.h Formats.hpp Registry.hpp Version.h
Config.hpp Device.hpp Logger.h Time.h Version.hpp
Constants.h Errors.h Logger.hpp Time.hpp
ConverterPrimitives.hpp Errors.hpp Modules.h Types.h
ConverterRegistry.hpp Formats.h Modules.hpp Types.hpp

/usr/local/include/SoapySDR:
Config.h Device.h Formats.hpp Registry.hpp Version.h
Config.hpp Device.hpp Logger.h Time.h Version.hpp
Constants.h Errors.h Logger.hpp Time.hpp
ConverterPrimitives.hpp Errors.hpp Modules.h Types.h
ConverterRegistry.hpp Formats.h Modules.hpp Types.hpp
q@q-X756UX:/opt/build/sdrangel/build$

q@q-X756UX:~$ SoapySDRUtil --find
######################################################
Soapy SDR -- the SDR abstraction library

######################################################
Found device 0
addr = 24607:1027
driver = remote
label = LimeNET-Micro [USB 2.0] 58399D024CE9
media = USB 2.0
module = FT601
name = LimeNET-Micro
remote = tcp://192.168.10.106:55132
remote:driver = lime
serial = 0058399D024CE9
Found device 1
driver = remote
label = SDRplay Dev0 RSP1 B0001P0002
remote = tcp://192.168.10.106:55132
remote:driver = sdrplay
serial = B0001P0002
q@q-X756UX:~$

q@q-X756UX:$ /opt/install/sdrangel/bin/sdrangel
2019-12-29 15:51:56.615 (I) SDRangel 4.12.4 Qt 5.9.5 64b x86_64 Ubuntu 18.04.3 LTS DSP Rx:24b Tx:16b PID 25067
2019-12-29 15:51:56.615 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/q/.config/f4exb/SDRangel.conf
2019-12-29 15:51:56.615 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #0: default
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #1: pulse
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #2: sysdefault:CARD=PCH
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #3: front:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #4: dmix:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #5: dsnoop:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #6: hw:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #7: plughw:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #8: alsa_input.pci-0000_00_1f.3.analog-stereo
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #9: alsa_output.pci-0000_00_1f.3.analog-stereo.monitor
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #0: default
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #1: pulse
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #2: sysdefault:CARD=PCH
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #3: front:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #4: surround21:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #5: surround40:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #6: surround41:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #7: surround50:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #8: surround51:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #9: surround71:CARD=PCH,DEV=0
[... 244 lines cut ...]
2019-12-29 15:52:00.041 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
2019-12-29 15:52:00.041 (D) MainWindow::MainWindow: end
2019-12-29 15:52:00.060 (D) GLSpectrum::initializeGL: context: major: 3 minor: 0 ES: no
2019-12-29 15:52:29.392 (D) MainWindow::sampleSourceChanged: tab at 0
2019-12-29 15:52:29.393 (D) DeviceAPI::saveSamplingDeviceSettings: not a suitable preset
2019-12-29 15:52:29.393 (D) DSPDeviceSourceEngine::stopAcquistion
2019-12-29 15:52:29.393 (D) DSPDeviceSourceEngine::handleSynchronousMessages: DSPAcquisitionStop
2019-12-29 15:52:29.393 (D) DSPDeviceSourceEngine::gotoIdle
2019-12-29 15:52:29.396 (D) FileInput::stop
2019-12-29 15:52:29.403 (D) SoapySDRInput::openDevice: allocated SampleFifo
2019-12-29 15:52:29.404 (D) SoapySDRInput::openDevice: open device here
2019-12-29 15:52:29.404 (D) DeviceSoapySDR::openopenSoapySDRFromSequence: driver=remote
2019-12-29 15:52:29.404 (D) DeviceSoapySDR::openopenSoapySDRFromSequence: serial=0058399D024CE9
[INFO] Make connection: 'LimeNET-Micro [USB 2.0] 58399D024CE9'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeNET-Micro
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M calibration values caching Disable
terminate called after throwing an instance of 'std::system_error'
what(): Invalid argument
Aborted (core dumped)
q@q-X756UX:$

Crash on selection of any of present remote devices, my guess is that trouble may
be on SDRAngel instalation side, the same issue opened on Soapy Remote git...

f4exb, 2019-12-29:
Then use Cubic SDR if it works fine for you. You do not need Soapy for LimeSDR that is natively supported.

9a4db, 2019-12-30:
Hi,
Here looking for fair ethernet network RX/TX use of LimeNET micro & RPI CM3 1GB RAM/ 8 GB on board EEPROM. 
Thank you for advice, but unfortunately so far Cubic is not able to TX.
Need Soapy Remote for that, in my case to allow other applications to be able to reach
LimeNET micro and SDRPlay RSP1 on LimeNET_USB port. 
I am long time user of your SW on Win10 and Ubuntu with LimeSDR USB on USB3 ports. 
Any way thank you for your great efforts on SDRAngel 👍

closed 2020-05-16 by f4exb