f4exb/sdrangel #461: DATV video stream using Pluto SDR

khan2018, 2019-12-30:
Hi

I am transmitting using Pluto SDR
[image]

and receiving using Pluto SDR
[image]

but unable to see any video stream. 
All checkboxes indicate the video is decoded.

Thanks

f4exb, 2019-12-31:
This is displayed in your second screenshot

khan2018, 2019-12-31:
I was using my android camera over the network using an app so DATV Express can take as an input. The video resolution said RGB in the end is it because of that I receive this green screen instead of the camera video ?

khan2018, 2019-12-31:
I want to save raw I/Q data and I use the toggle button to record.
[image]
Where is the output file being saved?

Thanks!

khan2018, 2019-12-31:
Is it possible to save .wav format ?

f4exb, 2020-01-02:
File is saved in the directory where sdrangel is run and has a `.sdriq` extension. This is raw 2x16 or 2x32 integer samples (depends on sample size) and has a 32 bytes header described here: https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/fileinput.

File name is in the form rec<device#>_<timestamp>.sdriq. For example: `rec0_2020-01-02T17_33_17_191.sdriq`

khan2018, 2020-01-03:
In windows 10, no such file is made in the installation directory...

closed 2020-02-08 by f4exb