f4exb/sdrangel #464: LimeSDR Mini not appearing as source

gcormier, 2020-01-03:
Ubuntu 18.04LTS, with LimeSDR packages from the PPA. udev rules installed.

dmesg
````
[ 2987.940056] usb 2-2: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[ 2987.960890] usb 2-2: New USB device found, idVendor=0403, idProduct=601f, bcdDevice= 0.00
[ 2987.960895] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2987.960899] usb 2-2: Product: LimeSDR Mini
[ 2987.960902] usb 2-2: Manufacturer: Lime Micro
[ 2987.960904] usb 2-2: SerialNumber: 1D..
````

SDRAngel installed from Snap, version 4.21.1, and I can see LimeSDR module installed. 

[image]

Output from running sdrangel that seems relevant
````
...
2020-01-03 10:46:48.095 (D) PluginManager::registerTxChannel  WFM Modulator  with channel name  "sdrangel.channeltx.modwfm"
2020-01-03 10:46:48.095 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fileinput
2020-01-03 10:46:48.095 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fcdpro
2020-01-03 10:46:48.095 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fcdproplus
2020-01-03 10:46:48.095 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.kiwisdrsource
2020-01-03 10:46:48.095 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.limesdr
2020-01-03 10:46:48.095 (D) DeviceLimeSDR::enumOriginDevices: Could not find any LimeSDR device
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.localinput
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.remoteinput
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.testsource
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.filesink
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.limesdr
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.localoutput
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.remoteoutput
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.testsink
2020-01-03 10:46:48.096 (D) DeviceEnumerator::enumerateMIMODevices: sdrangel.samplemimo.testmi
...
````

Another output
````
greg@biglenny:~/lime$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.1-myriadrf1~ubuntu18.04
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7
Search path:  /usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.7               (missing)
Search path:  /usr/local/lib/SoapySDR/modules0.7                                (missing)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libLMS7Support.so   (19.04.0)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librtlsdrSupport.so (0.3.0)
Available factories... lime, rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
greg@biglenny:~/lime$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  addr = 24607:1027
  driver = lime
  label = LimeSDR Mini [USB 3.0] 1D4..
  media = USB 3.0
  module = FT601
  name = LimeSDR Mini
  serial = 1D4..

````

gcormier, 2020-01-03:
Fun. It's a snap store issue, found this clicking around! Hopefully this issue helps someone down the road! Click permissions button and allow direct access to USB devices.

[image]

closed 2020-01-03 by gcormier

tdbone1, 2020-10-01:
> Fun. It's a snap store issue, found this clicking around! Hopefully this issue helps someone down the road! Click permissions button and allow direct access to USB devices.
> 
> [image]

Thanks this actually solved my problem. took me an hour to find this. thanks