f4exb/sdrangel #467: Version 5.0.0 compiler error QT5

loewal, 2020-01-18:
Hi,

Trying to compile v5.0.0 I got this error:

[ 48%] Built target filesource_autogen
[ 48%] Built target filesource
[ 48%] Automatic MOC and UIC for target remotesource
[ 48%] Built target remotesource_autogen
[ 48%] Building CXX object plugins/channeltx/remotesource/CMakeFiles/remotesource.dir/remotesource_autogen/mocs_compilation.cpp.o
In file included from /Users/loewal/Downloads/sdrangel-5.0.0/build/plugins/channeltx/remotesource/remotesource_autogen/mocs_compilation.cpp:2:
In file included from /Users/loewal/Downloads/sdrangel-5.0.0/build/plugins/channeltx/remotesource/remotesource_autogen/EWIEGA46WW/moc_remotesource.cpp:10:
In file included from /Users/loewal/Downloads/sdrangel-5.0.0/plugins/channeltx/remotesource/remotesource.h:21:
In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/QObject:1:
In file included from /opt/local/libexec/qt5/lib/QtCore.framework/Headers/qobject.h:46:
In file included from /usr/local/include/QtCore/qobjectdefs.h:48:
In file included from /usr/local/include/QtCore/qnamespace.h:43:
/usr/local/include/QtCore/qglobal.h:1044:9: error: assigning to 'QDebug::Stream *' from incompatible type 'long'
    t = std::forward<U>(newValue);
        ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/QtCore/qdebug.h:118:46: note: in instantiation of function template specialization 'qExchange<QDebug::Stream *, long>' requested here
    QDebug(QDebug &&other) noexcept : stream{qExchange(other.stream, nullptr)} {}
                                             ^
1 error generated.
make[2]: *** [plugins/channeltx/remotesource/CMakeFiles/remotesource.dir/remotesource_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [plugins/channeltx/remotesource/CMakeFiles/remotesource.dir/all] Error 2
make: *** [all] Error 2

Got any clue?
Thnx!

loewal, 2020-01-18:
Comparable when installing via Macports:
:info:build make[2]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build'
:info:build [ 47%] Built target modatvsrv
:info:build make[1]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build'
:info:build make: *** [all] Error 2
:info:build make: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build'
:info:build Command failed:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/work/build" && /usr/bin/make -j4 -w all VERBOSE=ON
:info:build Exit code: 2
:error:build Failed to build SDRangel: command execution failed
:debug:build Error code: CHILDSTATUS 4777 2
:debug:build Backtrace: command execution failed
:debug:build     while executing
:debug:build "system {*}$notty {*}$nice $fullcmdstring"
:debug:build     invoked from within
:debug:build "command_exec build"
:debug:build     (procedure "portbuild::build_main" line 8)
:debug:build     invoked from within
:debug:build "$procedure $targetname"
:error:build See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_SDRangel/SDRangel/main.log for details.

loewal, 2020-01-18:
Sorry,
Version 5.0.0 compiled with cmake & make and v4.12.5 with sudo macports outdated.
Same problem??

loewal, 2020-01-18:
Also tried in Parallels Desktop with Mint 19.

michaelld, 2020-01-27:
I see this too, in MacPorts and from source.

michaelld, 2020-01-27:
Temporary workaround proposed here: https://github.com/macports/macports-ports/pull/6245 ... just disable the part of SDRangel where this issue arises!

loewal, 2020-02-01:
A new error:

[ 54%] Linking CXX shared library ../../../lib/plugins/libbeamsteeringcwmod.dylib
Undefined symbols for architecture x86_64:
  "BeamSteeringCWMod::handleData(int, int)", referenced from:
      BeamSteeringCWMod::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) in mocs_compilation.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/plugins/libbeamsteeringcwmod.dylib] Error 1
make[1]: *** [plugins/channelmimo/beamsteeringcwmod/CMakeFiles/beamsteeringcwmod.dir/all] Error 2
make: *** [all] Error 2

loewal, 2020-02-01:
Sorry, who recognizes?

loewal, 2020-02-01:
Running on Macos Catalina  by the way

closed 2020-02-15 by loewal