f4exb/sdrangel #468: LimeSDR: LimeRFE support

f4exb, 2020-01-21:
This is to support LimeRFE from the LimeSDR plugin using I2C interface. Since it is linked to LimeSDR it opens possibilities like synchronizing band selection according to frequency (implemented in LimeSuite).

satfan52, 2020-02-05:
Today I tried to compile from scratch a new fresh installation following the instructions of the wiki,  I get the following error:

In file included from /opt/build/sdrangel/sdrbase/limerfe/limerfecontroller.cpp:20:0:
/opt/build/sdrangel/sdrbase/limerfe/limerfecontroller.h:23:10: fatal error: lime/limeRFE.h: No such file or directory
 #include "lime/limeRFE.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
sdrbase/CMakeFiles/sdrbase.dir/build.make:86: recipe for target 'sdrbase/CMakeFiles/sdrbase.dir/limerfe/limerfecontroller.cpp.o' failed
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/limerfe/limerfecontroller.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:559: recipe for target 'sdrbase/CMakeFiles/sdrbase.dir/all' failed
make[1]: *** [sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

I don't own a limesdr and I have not installed any limesdr directory in /opt/build so I wonder why I get that error.

f4exb, 2020-02-05:
Wrong version of LimeSuite

satfan52, 2020-02-05:
yes, this what I found in the cmake output
-- Found Lime Suite: /usr/local/include, /usr/local/lib/libLimeSuite.so
I have indeed found some leftover of previous (bad) installations and removed all libLimeSuite* files from both  /usr/local/include and  /usr/local/lib but it does not help, cmake keeps finding it ??? how is this possible

satfan52, 2020-02-05:
this is the content of /usr/local/include:
ad9361.h  ad9361-wrapper.h  wdsp.h

this is the content of /usr/local/lib/:
.                libad9361.so.0                   libmbe.a       libosc.so.HEAD-g7fa5158          librtlsdr.a          libserialdv.so.1.1.0  libxavna.la        osc        python3.7
..               libad9361.so.0.1                 libmbe.so      libosc.so.HEAD-g7fa5158-dirty    librtlsdr.so         libSoapySDR.so        libxavna.so        pkgconfig  python3.8
adi_colorimeter  libad9361.so.0.2                 libmbe.so.1    libosc.so.master-g6e76199        librtlsdr.so.0       libSoapySDR.so.0.8    libxavna.so.0      pypy2.7    R
cmake            libdsdcc.so                      libmbe.so.1.3  libosc.so.master-g6e76199-dirty  librtlsdr.so.0.6git  libSoapySDR.so.0.8.0  libxavna.so.0.0.0  python2.7  SoapySDR
fmc-tools        libgnuradio-osmosdr.so           libosc.so      libosc.so.master-g7fa5158        libserialdv.so       libwdsp.so            node_modules       python3.5  sysctl.d
libad9361.so     libgnuradio-osmosdr.so.0.1.5git  libosc.so.0    libosc.so.master-g9159e07-dirty  libserialdv.so.1     libxavna.a            ocaml              python3.6  systemd

satfan52, 2020-02-05:
ok problem solved, I have just cloned a fresh copy of sdrangel and it solved the issue after my cleanout of the lib directories

f4exb, 2021-08-07:
Abandoned

closed 2021-08-07 by f4exb