f4exb/sdrangel #469: Crash when adding Broadcast FM Demodulator to active sampling device.

zand, 2020-01-23:
Starting with 4abeeff adding the Broadcast FM Demodulator to an active rtl-sdr sampling device causes a Segmentation fault.

Reproduction:
1. Open sdrangel
2. Start rtl-sdr sampling device
3. Add a Broadcast FM Demodulator
4. Crash

Backtrace & GDB data from 664dc41:
```
2020-01-23 06:32:40.521 (W) AudioOutput::start: using system default device alsa_output.pci-0000_00_1b.0.analog-stereo
2020-01-23 06:32:40.521 (I) AudioOutput::start: audio format OK

Thread 11 "DSPDeviceSource" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff8d6ab700 (LWP 21185)]
0x00007ffff579b9a7 in QMutex::lock() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt full
#0  0x00007ffff579b9a7 in QMutex::lock() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff6fcda46 in MessageQueue::push(Message*, bool) (this=0xf03f, message=0x5555562f31c0, emitSignal=true)
    at sdrangel/sdrbase/util/messagequeue.cpp:46
#2  0x00007fffb53bba2f in BFMDemodBaseband::setBasebandSampleRate(int) (this=0x55555671a140, sampleRate=125000)
    at sdrangel/plugins/channelrx/demodbfm/bfmdemodbaseband.cpp:188
        msg = 0x5555562f31c0
#3  0x00007fffb53a32fc in BFMDemod::start() (this=0x55555666a110)
    at sdrangel/plugins/channelrx/demodbfm/bfmdemod.cpp:95
#4  0x00007ffff6f86570 in DSPDeviceSourceEngine::handleSynchronousMessages() (this=0x5555562df2c0)
    at sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:588
        sink = 0x55555666a110
        msg = 
          {<Message> = {_vptr.Message = 0x7ffff76ed4d8 <vtable for DSPSignalNotification+16>, static m_identifier = 0x0, m_destination = 0x0}, static m_identifier = 0x7ffff70385e8 "DSPSignalNotification", m_sampleRate = 125000, m_centerFrequency = 146400000}
        message = 0x7fffffffc860
#5  0x00007ffff7032679 in DSPDeviceSourceEngine::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x5555562df2c0, _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x5555565151d0)
    at sdrangel/build/sdrbase/sdrbase_autogen/JGTBRDHS57/moc_dspdevicesourceengine.cpp:80
        _t = 0x5555562df2c0
#6  0x00007ffff59b21b2 in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff670083c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff6708104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff59829c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff598513d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff59dc353 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff1413417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007ffff1413650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff14136dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff59db97f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff59809fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff579f23a in QThread::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff6f84a4d in DSPDeviceSourceEngine::run() (this=0x5555562df2c0)
    at sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:62
#19 0x00007ffff57a417d in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff2ccb6db in start_thread (arg=0x7fff8d6ab700) at pthread_create.c:463
        pd = 0x7fff8d6ab700
        now = <optimized out>
        unwind_buf = 
              {cancel_jmp_buf = {{jmp_buf = {140735565969152, -4974719339490206146, 140735565967104, 0, 93825006432960, 140737488344480, 4974548819043393086, 4974747538326265406}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#21 0x00007ffff4e8788f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
Currently compiling and running sdrangel on Ubuntu 18.04. Stopping the sample device before adding the Broadcast FM Demodulator then starting the sample device works fine. acd0892 build and runs without issue.

f4exb, 2020-01-25:
This is not limited to RTL-SDR

closed 2020-01-25 by f4exb

zand, 2020-01-26:
Would like take a moment to thank you for developing and maintaining such a great program.

Pulled and compiled 583a144 from scratch using a new build directory but unfortunately it still crashed with a similar back-trace. I checked and made sure the changes from c473c31 exist. Also tested using File Source device and got similar results as well.

commit 4d583406 (2020-01-25): BFM demod: fixed message to GUI initialization in baseband class. Fixes #469
commit c473c310 (2020-01-25): BFM demod: fixed message to GUI initialization in baseband class. Fixes #469