f4exb/sdrangel #470: Audio Input gets stuck after stopping and restarting transmission

ThomasTheGerman, 2020-01-28:
I set the Audio Input to either my microphone the pulseaudio monitor device, start transmitting, and get a clean signal that sounds great. 
Then I stop transmitting and start again, and the result is no proper signal at all, just a buzzing sound with regular clicking/popping. 
Disabling the audio input, switching around audio devices and starting/stopping transmission again gets rid of the problem, until I stop transmitting again.

If I first disable the audio input, then stop transmitting, the issue doesn't appear. Only if transmission is stopped while audio is being played back. 

Maybe it's an issue with the buffer still containing audio samples that haven't been transmitted yet, and it doesn't get cleared properly when the transmission is restarted? The noise is just a couple samples being repeated over and over. I've been able to reproduce it with different modulators and sinks selected.
[image]

System info: Arch, Pulseaudio 13.0, SDRAngel 4.13.0 compiled a couple hours ago.

f4exb, 2020-01-30:
For some reason this commit: https://github.com/f4exb/sdrangel/commit/583a144c67c71208c2edcd251904902ecd372c15 is causing more trouble than it is supposed to solve. It has to be fallen back.

f4exb, 2020-02-01:
Reverting the commit does not change anything. In fact the issue happens when the system default device is selected down in the list of devices. This device is marked with an "S" in the prefix column:

[image]

If you want to use the default device just leave the "System default device" selected at the top of the list which is in fact the default.

f4exb, 2020-05-03:
question has been answered

closed 2020-05-03 by f4exb