f4exb/sdrangel #473: Adapt to supported CPU extensions dynamically

FFY00, 2020-02-04:
See https://github.com/srsLTE/srsLTE/issues/449, same thing applies.

Currently, I maintain two packages in the Archlinux repos, one without optimizations and one with SSE 4.2.

f4exb, 2020-02-04:
"This problem is not hard to resolve. What we should do is compile the binary to the highest possible architecture and figure ut the supported CPU extensions at runtime."

No way...

closed 2020-02-04 by f4exb

FFY00, 2020-02-04:
Then I am droping it from the repos and recomending all the other maintainers to do the same.

I don't understand your reasoning, could you elaborate?

> No way...

This is already implemented for your by libvolk, you just need to replace your normal operations by libvolkn function calls.

FFY00, 2020-02-04:
> Note2: This "problem" is very hard to resolve and involves lots and lots of hard work and I have better things to do implementing new functionalities and solving real bugs.

No, you just need to use libvolk. I am not asking you to implement it from scratch, someone already did it for you.

f4exb, 2020-02-04:
It this is so easy then submit a PR and we will see.

FFY00, 2020-02-04:
I can add it to my TODO list however I am not sure when I will be able to do it as I have several upstream projects to maintain and I don't know sdrangel's codebase.

Could you please just have a look at the libvolk docs and consider it after?

f4exb, 2020-02-04:
I did and I maintain this is a pain and that I have better things to do
> Then I am droping it from the repos and recomending all the other maintainers to do the same.
No issues with this. I will not do any effort towards packaging. This is open source, you have the code please consider this as the real value.

FFY00, 2020-02-04:
> This "problem" is very hard to resolve and involves lots and lots of hard work and I have better things to do implementing new functionalities and solving real bugs.

I understand, but from my POV this just screams that you aren't really thinking about the users. Distributions are how users consume your software, by requiring them to compile it from source you are only making things hard on them and increasing the change of them breaking their system. Instaling things on the system from source isn't something supported by any distro that I know of, we won't provide support with your sdrangel installation. And one thing that confuses me is you providing debian packages, why are you doing this if users are supposed to compile from source?

FFY00, 2020-02-04:
> I did and I maintain this is a pain and that I have better things to do

I respect that, thanks. I won't bring it up again.

f4exb, 2020-02-04:
> but from my POV this just screams that you aren't really thinking about the users.

You have to consider that SDRangel is not a commercial product so if users cannot make the effort to compile the source while I make the effort to make the code public, resonably functioning and constantly implementing new things then they indeed do not deserve my attention

> Distributions are how users consume your software

Right you got it! This is not a mass consumption "product".

> increasing the change of them breaking their system

Sure..  by issuing `sudo make instal`l that every idiot on Linux would probably do. I have constantly warned about this. 

> And one thing that confuses me is you providing debian package

Consider that I do not want to break things that have been existing for a very log time and do not cost much anymore (same with Windows). I surely have considered dropping it but not definite on it still. I will carry on most of the new developments on the `v5` branch for which no packages exist.

FFY00, 2020-02-04:
> You have to consider that SDRangel is not a commercial product so if users cannot make the effort to compile the source while I make the effort to make the code public, resonably functioning and constantly implementing new things then they indeed do not deserve my attention

> Right you got it! This is not a mass consumption "product".

That is completely reasonable. I wish I had time to contribute to this project, as there are that many contributors, but unfortunately, I don't. In the future, when the project matures and is ready for mass consumption, and you want to start improving the user experience please let me know.

f4exb, 2020-02-05:
To elaborate a little bit more and show why libvolk will NEVER make it to SDRangel. I am currently dealing with LoRa and inspecting the code of gr-lora that uses libvolk. This is a code fragment that does the complex multiplication used for de-chirping the signal:

```c++
      // Dechirp the incoming signal
      volk_32fc_x2_multiply_32fc(down_block, in, &d_upchirp[0], d_num_symbols);

      if (d_state == S_READ_HEADER || d_state == S_READ_PAYLOAD)
      {
        volk_32fc_x2_multiply_32fc(up_block, in, &d_downchirp[d_offset], d_num_symbols);
      }
      else
      {
        volk_32fc_x2_multiply_32fc(up_block, in, &d_downchirp[0], d_num_symbols);
      }
```

compared to the same with plain c++ code:

```c++
    Complex dechirpUp = ci * m_downChirps[m_chirp]; // de-chirp the up ramp to get peamble and data
    Complex dechirpDown = ci * m_upChirps[m_chirp]; // de-chirp the down ramp to get sync
```

so not only this makes the code ugly and forces to explicit the data size but it also means changing thousands lines of code to implement libvolk throughout SDRangel. Of course the code you see just above is part of a loop but any modern compiler (g++, clang) takes care of vector optimizing the code for me having the code still pleasant to read. Code is not just there to make the job it has to be as elegant as possible to be maintainable. As stated earlier code IS the value. Using libvolk is a step back in technology.

f4exb, 2020-02-21:
This is it: binaries have been removed from the releases section.

FFY00, 2020-03-23:
I was looking for alternatives and GCC's `target_clones` attribute seems interesting. Do you think it makes sense to explore that?

https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html#index-target_005fclones-function-attribute