f4exb/sdrangel #474: question: can source/sink file plugins process real time I/Q sample of a custom board

satfan52, 2020-02-05:
I am participating to a new sdr project which that is still in its infancy and in its design pĥase (so no github repository yet). I was wondering if for the initial tests of that custom board the so called source/sink file plugins could be use to allow that custom board to work with  sdrangel (provided of course that the format of the data is right and the right protocol is followed). Is for that purpose or are these source/sink pluggins designed only to replay already recorded .sdriq files.  If not what other gneric plusgins xould be used for making the initial tests ?  remote input/output plugins ? The custom board will be connected over USB to the PC and will be using a FGPA

f4exb, 2020-02-05:
File plugins (devices and channels) are meant to be used with .sdriq files that have a specific header. You can possibly mimic this and use named pipes but this looks a bit complex/dodgy when you consider you have to synchronize things (it is not simply streaming randomly because of this header)

Remote plugins (devices and channels) are meant to make a connection between SDRangel instances over the network and use a specific protocol with optional FEC to handle UDP communication over a lossy channel. Implementing this in a FPGA could be possible since there are now big FPGAs that can synthesize MCUs but this is not what I would qualify as "the simpler the better"

The UDP source (channel) is maybe what you are looking for as it takes its input from a UDP connection with simple interleaved I/Q samples over UDP. However this is meant to be used with a transmitter. This also looks a bit complex but complexity is not on the communication side with your device. You also have the File output plugin that dumps the baseband into a .sdriq file or the Test output that will display the baseband in a small spectrogram.

There is not handling of plain UDP streams at device level (UDP input / output plugins) at the moment.

satfan52, 2020-02-05:
Ok so if there is no handling of plain UDP streams at device level (UDP input / output plugins) then what is next in line in the path of least resistance to succeed to use SdrAngel with a custom sdr board without software changes to SdrAngel ? use SoapySdr device plugin with a Q&D custom module? or there is just realistically no short cuts ?

f4exb, 2020-02-06:
Hello, if you create a module for Soapy then this is possible to use the SoapySDR plugin without changes.

satfan52, 2020-02-06:
I guess there is then no reference design that you could recommend to follow as the module sits  between the hardware and SoapySDR with no direct nterface with sdrangel.

f4exb, 2020-02-07:
Yes check on Soapy side a lot of hardware was ported to Soapy.

closed 2020-02-10 by satfan52