f4exb/sdrangel #475: sdrangel doesn't try to scan for soapysdr devices

OrthoCube, 2020-02-05:
I am running Manjaro Linux (based on Arch Linux) and installing `libsdrplay`, `soapysdr` and `soapysdrplay` and finally `sdrangel`, my SDRplay RSP1A is not detected at all. Even if I connect my RTL-SDR, it does show up but not as a SoapySDR device (I believe it should be listed twice, as a generic DTV device and as a SoapySDR device.

Checking logs while soapysdr is launching, it doesn't mention aything about scanning for SoapySDR devices, but `SoapySDRUtil` can properly detect my SDRplay RSP1A.

Can anyone please give me suggestions on how to make sdrangel detect SoapySDR devices? Or do I really have to build from source?

Thank you in advance.

f4exb, 2020-02-07:
If the package is build with external libraries option then Soapy is not included. This is probably the case and this is definitely the case of the .deb files you will find in the releases section. So yes generally to get Soapy support you need to build from source.

f4exb, 2020-02-08:
See #472

closed 2020-02-08 by f4exb