f4exb/sdrangel #476: Couldn't install Pluto plugin.

luozhen731, 2020-02-05:
Hi,

I tried to follow the steps (https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux) to install plutoSDR plugin into SDRangel.

1. install iio
2. install Non-hardware dependencies
3. install Soapy SDR
4. install Soapy Remote
5. Build SDRangel

 However, I could not find it from sampling devices list.
[image]
 Do I miss something during the build? Could some one help me on this?

Kind regards,
Zhen

satfan52, 2020-02-05:
Hello, just in case it helps, I also follow the same instructions  for compiling under linux and my pluto is detected without any issues. 

It is actually detected twice ;-) directly and also  asecond time as a SoapySDR device because I have installed SoapySDR and the SoapyPlutoSDR module (for soapysdr testing purposes only)

I noticed that you mention that you install iio at the beginning of the procedure, that is not what is specified in the instructions. So I would try uninstalling iio it completely  and then reinstalling iio as indicated precisely in the instructions for the plutosdr under the "hardware dependencies".  You may need to re-clone a fresh copy of sdrangel after you uninstall iio. 

cd /opt/build
git clone https://github.com/analogdevicesinc/libiio.git
cd libiio
git reset --hard 826563e41b5ce9890b75506f672017de8d76d52d
mkdir build; cd build
cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/libiio -DINSTALL_UDEV_RULE=OFF ..
make -j4 install

f4exb, 2020-02-07:
Hello, this should not normally have an impact as long as iio is installed before you run SDRangel cmake (so it can discover it). However as you may have noticed the udev rules are not installed automatically. You must make sure these are installed to be able to access see: https://github.com/f4exb/sdrangel-docker/tree/master/udev-rules

luozhen731, 2020-02-10:
> Hello, this should not normally have an impact as long as iio is installed before you run SDRangel cmake (so it can discover it). However as you may have noticed the udev rules are not installed automatically. You must make sure these are installed to be able to enumerate the Pluto see: https://github.com/f4exb/sdrangel-docker/tree/master/udev-rules

Hello,
I install with dvb file and after some time (the plugin doesn't show at beginning but after several restart and re-power Pluto) the plugin shows out. Thank you anyway for your reply. I do didn't notice this udev-rules. Perhaps this is the reason.

f4exb, 2020-02-12:
Added this paragraph in the quick start Wiki: https://github.com/f4exb/sdrangel/wiki/Quick-start#linux-udev-rules

luozhen731, 2020-02-12:
> Added this paragraph in the quick start Wiki: https://github.com/f4exb/sdrangel/wiki/Quick-start#linux-udev-rules
> 
> Also if your hardware is physically not accessible it will not show up either this may happen with a faulty cable as an intermittent issue may indicate.

Thank you so much.
Kind regards,
Zhen

closed 2020-02-12 by luozhen731