f4exb/sdrangel #477: Segmentation fault

drom, 2020-02-05:
```
drom@linux-sydi:~> sdrangel
2020-02-05 13:19:10.994 (W) Attribute Qt::AA_EnableHighDpiScaling must be set before QCoreApplication is created.
2020-02-05 13:19:10.995 (I) SDRangel 5.1.0 Qt 5.14.1 64b x86_64 openSUSE Tumbleweed DSP Rx:24b Tx:16b PID 6488
2020-02-05 13:19:10.995 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/drom/.config/f4exb/SDRangel.conf
2020-02-05 13:19:11.467 (W) qUncompress: Input data is corrupted
Segmentation fault (core dumped)
```

drom, 2020-02-09:
Segfault happens in v5.0.0 and v5.1.0
Do you need more details to reproduce?

f4exb, 2020-02-12:
You bet...

Supposedly you have compiled from source since there are no (and there will never be any) binaries released (or at least any that I should care of) from the v5 branch.  The reference environment is Ubuntu 18.04 so I am afraid I am not able to check anything if not compiled in this environment. The piece of log shows it is running in OpenSUSE Tumbleweed.

However if you compile in debug mode and provide a stack trace I could possibly have hints about what may be going wrong. You may also try again in a VM with Ubuntu 18.04 and see what happens.

drom, 2020-02-12:
I am getting OpenSuse RPM from community repo:
https://download.opensuse.org/repositories/hardware:/sdr/openSUSE_Tumbleweed/x86_64/sdrangel-5.1.0-1.2.x86_64.rpm.mirrorlist
It was working fine before v5
I will try finding out people behind the build.

drom, 2020-02-12:
@f4exb can it be because of Qt 5.14.1 ?

mnhauke, 2020-02-12:
I'm the maintainer of the mentioned openSUSE package on OBS.

The issue is related to gcc  LTO that got enabled in openSUSE Tumbleweed  some time ago...
https://lists.opensuse.org/opensuse-factory/2019-07/msg00240.html
https://gcc.gnu.org/wiki/LinkTimeOptimization

This means the distro provided CXXFLAGS  changed from
```
CXXFLAGS='-O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection  -g'
```
to
```
CXXFLAGS="-O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -flto=4 -g -Werror=implicit -Werror=maybe-uninitialized -Werror=memset-transposed-args -Werror=null-dereference -Werror=sizeof-array-argument -Werror=sizeof-pointer-memaccess" 
```
Interestingly this wasn't an issue with SDRangel < 5.0.0.

It probably makes sense search for the root cause of this problem since other Linux distributions also considering to use this gcc feature.

I just have submitted an update to the SDRangel package on OBSy were I have disabled LTO.

drom, 2020-02-13:
@mnhauke I will be waiting for an update package to appear in the repo.

drom, 2020-02-13:
@f4exb here are some more information from `dmesg -w`:

```
[14893.798442] sdrangel[4720]: segfault at 0 ip 00007ff92c1b96dd sp 00007ffe47abb900 error 4 in libbeamsteeringcwmod.so[7ff92c1b7000+d000]
[14893.798446] Code: 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 41 54 55 48 83 ec 08 4c 8b 25 ba 18 01 00 48 8b 2d e3 18 01 00 49 8b 14 24 48 89 55 00 <8b> 02 83 c0 01 83 f8 01 0f 87 be 00 00 00 48 8d 35 06 aa 00 00 48
```

f4exb, 2020-02-13:
Since v5 I am only supporting Ubuntu 18.04 and compilation from source period. I have no time dealing with build issues on other environments other than my current environment. v5 is not the default branhc (master) and if people start to use it to build packages I think I will consider ending SDRangel to fork out active development to another project in a private repo.

closed 2020-02-13 by f4exb

drom, 2020-02-13:
Don't get me wrong. I am just a loyal user of SDRangel on OpenSuSE Tumbleweed. I love the project. Was trying to help.

f4exb, 2020-02-13:
Sorry I am not considering issues on v5 if not compiled from source in Ubuntu 18,04

f4exb, 2020-02-14:
> It probably makes sense search for the root cause of this problem since other Linux distributions also considering to use this gcc feature.

Sorry there is no consensus on this. Addressing the "root cause" means adding the `volatile` keyword all over the place even for no obvious reasons so this task is almost impossible. There is also the proposal (not consensual either) to simply deprecate the `volatile` keyword in C++. Having said that the only reasonable action is just wait and see.