f4exb/sdrangel #479: Could not start sample source

magellan-13016, 2020-02-18:
Hi,
I'm David and I just discovered this software today. I'm using Linux Mint 19.3 and genuine RTL-SDR RTL2832U v3. All is OK with GQRX but I can't start it with SDRAngel.

I always get the error "Could not start sample source".

I have done the "export LD_LIBRARY_PATH=/usr/lib/sdrangel" command before but there is no way to start the dongle.

I'm using the last stable release : sdrangel_4.13.0-1_amd64.deb

Could you help me ?

Thanks, David.

magellan-13016, 2020-02-21:
Solved ! Not working on Linux Mint 19.3 and my SDR Key. That's all.

f4exb, 2020-02-21:
Due to the unwillingness of the logger to provide useful information other than "it does not work" this issue can indeed be closed.

closed 2020-02-21 by f4exb

f4exb, 2020-02-23:
We had a very hot debate with the logger and now he banned my address from his mailing system so no direct contact is possible anymore. However on the positive side and for the larger public this made me reconsider a few things. I recognize my responsibility in what was not made clear. So I will consider these action points at the earliest:
  
  - make it clear what I consider as the fair use of this software
  - make it clear what is expected when posting issues
  - reconsider the distribution of binary artifacts. They tend to make you think that things are given for granted (I have this inclination too so I know it well). This is definitely what is NOT meant by Open Source (TBD in the "fair use" listed above):
    - remove artifacts from the releases section (done).
    - at least to be nice to the loyal users put it place a CI/CD system to build artifacts so that the build procedure is made clear, controlled and open to contributions

k2extreme, 2022-09-15:
Hello,
I have similar problem but the problem appear when I'm trying to use two RTL-SDR RTL2832U v3 simultaneously. OS: Fedora 36. Edouard would you consider to solve the problem?

f4exb, 2022-09-15:
you probably have the same serial ids for both. You need to change it with
rtl-sdr tools

k2extreme, 2022-09-15:
Thank you very much for your help. I considered something wrong with ID's of dongles. So I will try the solution. Thank you. I will post info about result.

k2extreme, 2022-09-15:
The solution is working:
Linux terminal command to define SN of RTL-SDR is:

rtl_eeprom -s nnnn 

this command works if only one of devices is connected to PC.
Thank you :)

maximilien1990, 2024-02-26:
i have the same problem on arch with my hackrf. any suggestions?