f4exb/sdrangel #484: Ubuntu 18.04 snap install.

oh3gai, 2020-03-01:
Hi! 

First of all awesome work!

I'm not pro with linux and I faced problem with Ubuntu 18.04 LTS snap install method with hackrf. The latest snap intall of Sdrangel starts like charm but hackrf is missing from source devices list. I'm able to see my connected hackrf without sudoing from bash. So I suppose the user rights to usb devices are ok. Does anybody else have this problem? am I doing something wrong?

I tryed to check inside snap install folders and i couldn't find hackrf libraries.

I love the idea of snap install. It just for my skils..  sorry if this is false alarm.. 

Thanks for great work!!

Br.  Oh3gai / Jarno

f4exb, 2020-03-01:
Snap is not supported. If you cannot or do not want to compile from source or use Docker please use the Debian package.

oh3gai, 2020-03-01:
Hi! 

Thanks for fast reply! I found the deb file and tried the dpkg -i  aproach . Worked !!

Thanks for help!

Br,Jarno

closed 2020-03-01 by oh3gai