f4exb/sdrangel #487: SDRAngel UDP doesn't work with redsea

vadimostanin, 2020-03-12:
SDRAngel UDP audio output doesn't work with redsea(https://github.com/windytan/redsea)
Preference->Audio
    audio rate is 170k, (Left/Right), decimation is 1, 16-bit
nc -l -u 9998 | redsea -r 170k

f4exb, 2020-03-14:
This cannot work with audio (Preference->Audio). Redsea needs the raw FM demodulation so you have to use the [UDP Sink plugin](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/udpsink/readme.md) instead and not piping the audio out of the BFM demod.

In (6) choose NFM mono, It says "N" for narrow but the bandwidth can be selected next (7) and the FM deviation with (8). You have to tweak FM deviation but I suppose 50000 Hz is a  good start. 

Note that the size of samples depends on the choice in the second combo box in (6). It is either:
  - 16 bits: samples are 16 bit signed with little endian layout (S16LE) 
  - 24 bits: samples are 32 bit signed with little endian layout (S32LE) using only the 3 less significant bytes. This means that the range is -2²³ to 2²³ - 1

Hence some conversion may be required before piping into redsea. Possibly some more formats could be added as an enhancement.

f4exb, 2020-04-04:
Question has been answered

closed 2020-04-04 by f4exb