f4exb/sdrangel #488: Broadcast FM - segmentation fault

makkiato83, 2020-03-16:
Hello,

I have just installed sdrangel, in Ubutu 18.02, following the instructions on the wiki.
My Hardware is a HackRF.

When I start the program, as select HackRF as input (2.4Msamp/s), with DC-correction and IQ inversion (LSB-->USB, USB<--LSB). The waterfall displays correctly the incoming signal. 

If I add a AM-demodulator, it works fine.

However if I add a Broadcast FM demodulator, I get the crash below.
Is there anything easy I can do to avoid this crash?

PS: I have also tried with different sample rates (e.g., 10Msamp/sec).

2020-03-16 13:56:58.301 (D) LowPassFilterRC::configure: t:  2.4  a1:  -0.659241  b0:  0.340759
2020-03-16 13:56:58.301 (D) LowPassFilterRC::configure: t:  2.4  a1:  -0.659241  b0:  0.340759
2020-03-16 13:56:58.301 (D) BFMDemodSink::applySettings: MsgConfigureBFMDemod:  m_inputFrequencyOffset:  0  m_rfBandwidth:  180000  m_afBandwidth:  15000  m_volume:  2  m_squelch:  -60  m_audioStereo:  false  m_lsbStereo:  false  m_showPilot:  false  m_rdsActive:  false  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  force:  true
2020-03-16 13:56:58.301 (D) PhaseLock::configure: freq: 0.395833 bandwidth: 0.001042 minsignal: 0.010000
2020-03-16 13:56:58.301 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.301 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.302 (D) Interpolator::createPolyphaseLowPass: ntaps: 18
2020-03-16 13:56:58.302 (D) BFMDemodSink::applyChannelSettings:  channelSampleRate:  48000  channelFrequencyOffset:  0
2020-03-16 13:56:58.302 (D) NCO freq: 0.000000 phase inc 0
2020-03-16 13:56:58.302 (D) PhaseLock::configure: freq: 0.395833 bandwidth: 0.001042 minsignal: 0.010000
2020-03-16 13:56:58.302 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.302 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.302 (D) Interpolator::createPolyphaseLowPass: ntaps: 18
2020-03-16 13:56:58.303 (D) BFMDemodBaseband::BFMDemodBaseband
2020-03-16 13:56:58.303 (D) AudioDeviceManager::addAudioSink: -1: 0x5582e2b90740
2020-03-16 13:56:58.303 (W) AudioOutput::start: using system default device alsa_output.pci-0000_00_1b.0.analog-stereo
2020-03-16 13:56:58.303 (I) AudioOutput::start: audio format OK
2020-03-16 13:56:58.303 (D) AudioOpus::AudioOpus: libopus version libopus 1.1.2
2020-03-16 13:56:58.303 (D) RTPSink::RTPSink: initialized transmitter: No error
2020-03-16 13:56:58.303 (D) RTPSink::RTPSink: created transmitter: No error
2020-03-16 13:56:58.303 (D) RTPPacketBuilder::CreateNewSSRC: timestamp: 3422946714
2020-03-16 13:56:58.303 (D) RTPSink::RTPSink: created session: No error
2020-03-16 13:56:58.303 (D) RTPSink::setPayloadInformation: payloadType: 0 sampleRate: 48000
2020-03-16 13:56:58.303 (D) RTPSink::setPayloadInformation: timestamp unit set to 0.000021: No error
2020-03-16 13:56:58.303 (D) RTPSink::setPayloadInformation: set default mark to false: No error
2020-03-16 13:56:58.303 (D) RTPSink::setPayloadInformation: set default timestamp increment to 960: No error
2020-03-16 13:56:58.303 (D) RTPSink::setPayloadInformation: set maximum packet size to 1940 bytes: No error
2020-03-16 13:56:58.306 (D) BFMDemodSink::applyAudioSampleRate: 48000
2020-03-16 13:56:58.306 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.306 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.306 (D) LowPassFilterRC::configure: t:  2.4  a1:  -0.659241  b0:  0.340759
2020-03-16 13:56:58.306 (D) LowPassFilterRC::configure: t:  2.4  a1:  -0.659241  b0:  0.340759
2020-03-16 13:56:58.306 (D) BFMDemod::applySettings: MsgConfigureBFMDemod:  m_inputFrequencyOffset:  0  m_rfBandwidth:  180000  m_afBandwidth:  15000  m_volume:  2  m_squelch:  -60  m_audioStereo:  false  m_lsbStereo:  false  m_showPilot:  false  m_rdsActive:  false  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  force:  true
2020-03-16 13:56:58.306 (D) DSPDeviceSourceEngine::addSink:  BFMDemod
2020-03-16 13:56:58.306 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2020-03-16 13:56:58.306 (D) BFMDemod::handleMessage: DSPSignalNotification
2020-03-16 13:56:58.306 (D) BFMDemod::start
2020-03-16 13:56:58.306 (D) DownChannelizer::applyChannelization done:  in (baseband): 2400000  req: 0  out (channel): 2400000  fc: 0
2020-03-16 13:56:58.306 (D) BFMDemodSink::applyChannelSettings:  channelSampleRate:  2400000  channelFrequencyOffset:  0
2020-03-16 13:56:58.306 (D) NCO freq: 0.000000 phase inc 0
2020-03-16 13:56:58.306 (D) PhaseLock::configure: freq: 0.007917 bandwidth: 0.000021 minsignal: 0.010000
2020-03-16 13:56:58.306 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.306 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2020-03-16 13:56:58.306 (D) Interpolator::createPolyphaseLowPass: ntaps: 18
Segmentation fault (core dumped)

f4exb, 2020-03-16:
Failry easy to reproduce. The device must be running while adding the BFM demod:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f6013f5f9a7 in QMutex::lock() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
[Current thread is 1 (Thread 0x7f5f9f7fe700 (LWP 15292))]
(gdb) bt
#0  0x00007f6013f5f9a7 in QMutex::lock() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007f60157c45b0 in MessageQueue::push(Message*, bool) (this=0x7e0, message=0x7f5f94005c40, emitSignal=true) at /opt/build/sdrangel/sdrbase/util/messagequeue.cpp:46
#2  0x00007f5fd9a99f1b in BFMDemodBaseband::setBasebandSampleRate(int) (this=0x561e2ac4fe10, sampleRate=250000) at /opt/build/sdrangel/plugins/channelrx/demodbfm/bfmdemodbaseband.cpp:188
#3  0x00007f5fd9a80a2d in BFMDemod::start() (this=0x561e2ada5d10) at /opt/build/sdrangel/plugins/channelrx/demodbfm/bfmdemod.cpp:95
#4  0x00007f6015770a52 in DSPDeviceSourceEngine::handleSynchronousMessages() (this=0x561e29e0cdf0) at /opt/build/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:588
#5  0x00007f6015832af1 in DSPDeviceSourceEngine::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x561e29e0cdf0, _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x561e2a82ceb0)
    at /opt/build/sdrangel/build.debug/sdrbase/sdrbase_autogen/JGTBRDHS57/moc_dspdevicesourceengine.cpp:80
#6  0x00007f60141760c2 in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007f6014ec483c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007f6014ecc104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007f60141468d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007f601414904d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007f60141a0263 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007f600f8dc417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007f600f8dc650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007f600f8dc6dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007f601419f88f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007f601414490a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007f6013f6323a in QThread::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007f601576e01a in DSPDeviceSourceEngine::run() (this=0x561e29e0cdf0) at /opt/build/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:62
#19 0x00007f6013f6817d in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007f60111946db in start_thread (arg=0x7f5f9f7fe700) at pthread_create.c:463
#21 0x00007f601364b88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
Strangely enough this should have already been fixed with #469

makkiato83, 2020-03-17:
Thanks. Indeed activating the BFM *before* starting the HackRF everything works fine.

Fantastic software btw, I have just started discovering it. Keep up the great work!

f4exb, 2020-03-25:
To be retrofitted in v4

f4exb, 2020-03-29:
Implemented in v5.3.2 and v4.14.2

closed 2020-03-29 by f4exb