f4exb/sdrangel #491: Capitals in filenames 'conflict'

loewal, 2020-03-22:
Hi,
Because I got a compilererror with sdrangel 5.3.1 I contacted the maker of Keka, the archiveprogram for the Mac. When I decompressed the zip zourcodefile with the builtin archiver, no problem...
So this is the answer of the Keka programmer:

Now I see. @loewal the problem here is that the source code has indeed two versions of the file, it must be in a case sensitive file system and probably @f4exb didn't noticed it.

The macOS bundled extractor just overwrites the lower case SWGDevicesetApi into SWGDeviceSetApi, because the later is extracted first. This is even worst because the lower case is an outdated file with version 5.1.0 while the upper case is version 5.3.0. Keka notes the difference and deppending on the file system you're extracting to it renames the lower case to SWGDevicesetApi_1 or respects the two versions.

The older version 5.1.0 has the class declared in lower case SWGDevicesetApi and I suppose the rest of the code is linked to this class name, that's why the compiler asks for it. With Keka the newer version file 5.3.0 is used but there the class is defined with upper case SWGDeviceSetApi, so the compiler fails.

See here the offending files:

SWGDeviceSetApi.h (upper case latest 5.3.0 version)
SWGDeviceSetApi.cpp (upper case latest 5.3.0 version)
SWGDevicesetApi.h (lower case 5.1.0 version)
SWGDevicesetApi.cpp (lower case 5.1.0 version)
Listing of the ZIP contents (two first/last selected files):

Screen Shot 2020-03-22 at 10 44 53

—
You are receiving this because you were mentioned.
Reply to this email directly, view it on GitHub, or unsubscribe.

f4exb, 2020-03-25:
Maybe it is a mishap of the code generator the ones that should not be there are: SWGDevicesetApi.*

f4exb, 2020-03-29:
Implemented in v5.3.2 and v4.14.2

closed 2020-03-29 by f4exb